supermerill/SuperSlicer #2720: Superslicer crashes on clicking 'Add/Remove filaments' or 'Add/Remove Presets'

Herjanz, 2022-04-17:
### Version
Version 2.3.57.12

### Operating system type + version
Arch Linux 5.17.3.arch1-1
X server, Budgie (Gnome)

### Behavior
Application crashes/terminates when I click (with either left or right mouse button) "Add/Remove Filaments" or "Add/Remove Presets" on top right. This happens in all simple/advanced/expert modes.

journalctl -r -n 40 gives: [https://pastebin.com/6VSx46JQ](https://pastebin.com/6VSx46JQ) 
'Add physical printer' works though. And in PrusaSlicer the 'Add/Remove' features still work.

Herjanz, 2022-04-23:
@supermerill it actually does crash!

Dialog shown:
Title: SuperSlicer error
Msg: Internal error: boost::filesystem::directory_iterator::construct: No such file or directory [system:2]: "/usr/share/SuperSlicer/profiles"
Button: OK

on clicking "OK" application terminates

Herjanz, 2022-04-23:
I added a new folder: sudo mkdir /usr/share/SuperSlicer/profiles
And now it doesn´t crash anymore.

Herjanz, 2022-04-28:
@supermerill No, I do not have any presets. I thought it was weird since in PrusaSlicer I had a lot.

supermerill, 2022-07-13:
I think it's better to uncompress the .tar instead of using the the appimage on your system.

supermerill, 2024-05-21:
can't reproduce on ubuntu 22.04
still an issue?

closed 2024-05-22 by Herjanz