supermerill/SuperSlicer #2721: Seam Sphere ignored

alexvonsiebenthal, 2022-04-17:
### What happened?

Upgraded to 2.4.58.3
Now getting random seams despite setting a seam sphere
[image]

### Project file & How to reproduce

[SeamSphereBug.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8502265/SeamSphereBug.3mf.zip)

### Version

2.4.58.3

### Operating system

macOS 12.3

### Printer model

voron 2.4 / stealthburner

alexvonsiebenthal, 2022-04-17:
Additionally (maybe worth another issue?), I cannot set any seam settings anymore: 
[image]

alexvonsiebenthal, 2022-04-19:
@supermerill any update?

hagan66, 2022-04-20:
I noticed this too, and it keeps me from using the 2.4 version.

supermerill, 2022-04-22:
fixed in latest nightly

alexvonsiebenthal, 2022-04-27:
Are the nightlies pre-compiled? If so, where can they be found?

neophyl, 2022-04-28:
https://github.com/supermerill/SuperSlicer/actions  then select one of the builds for your system on the left.

For example for Windows https://github.com/supermerill/SuperSlicer/actions/workflows/ccpp_win.yml

closed 2024-02-11 by supermerill