supermerill/SuperSlicer #2723: variable_layer_height not considered in G_code generation

Ygle38, 2022-04-18:
### Version
2.3.57.12

### Operating system type + version
Windows 7 SP1 64-bit

### Behavior
* variable_layer_height activated in printer sheet, variable height selected and tuned in 3D view (visualized properly in the vertical right bar), but not considered in Slicer sheet and G-code view sheet (keeping using first layer thickness and regular fixed thickness for other layers). G-code export and analysis confirm there is no variable height computation.

_Is this a new feature request?_
* No

#### Project File (.3MF) where problem occurs
[Bug_variable_height_Ygle38.zip](https://github.com/supermerill/SuperSlicer/files/8506636/Bug_variable_height_Ygle38.zip)
_

Super-Matter, 2022-04-18:
https://github.com/supermerill/SuperSlicer/issues/2714

I had basically the same issue, check it out

Ygle38, 2022-04-18:
Hello, I can see the same issue on 2.3 branch (Pre-release 2.3.57.12) already reported, sorry for spaming: #2652

Ygle38, 2022-04-18:
Bug considered and almost understood in #2113  ;-)
I do close this thread.

closed 2022-04-18 by Ygle38