supermerill/SuperSlicer #2727: SuperSlicer adding accelerations, jerk, and feedrates to gcode file printer cannot handle

maverickccie, 2022-04-18:
### What happened?

SuperSlicer is adding the following lines to start of every Gcode file for every printer:

M201 X5000 Y5000 Z500 E5000 ; sets maximum accelerations, mm/sec^2
M203 X1440000 Y1440000 Z54000 E216000 I0; sets maximum feedrates, mm/min
M204 P1500 T1500 ; sets acceleration (P, T), mm/sec^2
M566 X60000.00 Y60000.00 Z1200.00 E1800.00 ; sets the jerk limits, mm/min

Although I have one printer that can do this, it still messes with the bed leveling to move that fast. I cannot find anywhere that I set anything to these kind of speeds.

This started after I used the bridge flow calibration tool in 2.3.57.11.

### Project file & How to reproduce

Installed SuperSlicer on another computer
Setup Printer from scratch with all the same settings (same for print and filament settings)
Use bridge flow calibration tool
slice
Gcode file has setting in it.
All Gcode files after that have setting in them

Wiped out SuperSlicer
Setup Printer from scratch with all the same settings (same for print and filament settings)
Did not use any of the calibration tools
None of the Gcode files have the bad settings in them.

### Version

2.3.57.11

### Operating system

Windows 10

### Printer model

Railcore II ZLT, Railcore II ZL, Prusa MK2 (Highly modified), Prusa MK3S, DBOT, FT-5(Highly modified)

mmirate, 2022-04-19:
Like the g-code comment says, it is outputting millimeters _per minute_. G0/G1 feedrates are all in that unit, and some firmware uses that unit for speedlimits too. But in the slicer, the settings are accepted in millimeters _per second_. Double-check that your slicer settings are in the correct units, and that your G-Code Flavor agrees with what your machine is running.

supermerill, 2022-04-22:
it's the limits from the printer limits, when set to "also emit limits to gcode"
If you find some problem with unmodified built-in profile, tell me which one.