supermerill/SuperSlicer #2728: Uploading files to octoprint not working despite test button says connection is working

ThaMighty90, 2022-04-19:
### Version
2.4.58.2

### Operating system type + version
Windows 11

### Behavior
When setting up direct upload to Octoprint, the Test button states that the connection is working. When
trying to upload gcode it fails with HTTP-404

[image]

[image]

In the previous version 2.3.57.12 uploading works without any problems.

#### Project File (.3MF) where problem occurs
Happens with every upload, no matter what the 3d model is.

supermerill, 2022-04-29:
I'll test

supermerill, 2022-05-03:
the test call `http://localhost/api/version`
and the upload  `http://localhost/api/files/local`

your octoprint server is listening to `http://localhost/api/files/local` ?

ThaMighty90, 2022-05-04:
i think so, i haven't changed anything regarding the behaviour in Octoprint. As I already mentioned with the previous version of SuperSlicer the uploads work without any problems, it is just the new version that gives me trouble. I can access http://localhost/api/files/local from the browser but get a permission denied because i can't pass the API Key through my broswer.

supermerill, 2022-05-05:
Can you run SuperSlicer with the `--loglevel=5` option?
If you create a shortcut of superslicer_console.exe, just add it after the target
[image]

ThaMighty90, 2022-05-05:
Okay, I started the superslicer_console.exe with loglevel 5 and had a look at the output. What I noticed is that you connect to the URL to check the Octoprint Version which is working but than you resolve the URL to its IP. In my case that can't work because my network setup needs the URL. The IP Address doesn't work on Port 80 because it gets all routed via a reverse proxy because i have many different things that would listen on port 80. In Version 2.3.57.12 you used the URL which worked.

This is Version 2.3.57.12
[image]

This is Version 2.4.58.2
[image]

The IP Resolve is what makes it impossible on my setup to find the Octoprint server by SuperSlicer. The routing in my network which service you want to get on the server all depends on the URL and not on IP:Port

supermerill, 2022-05-05:
np, that option is completely undocumented and hidden in the conf file.