supermerill/SuperSlicer #2729: Variable fan speed by layer height/modifier/separate part settings

xabixpf, 2022-04-20:
The greatest drawback and due to which I still prefer sometimes to use Simplify3D is that Prusaslicer/Superslicer still have not implemented fan speed control by means of  separate object settings, layer ratio or modifier. I vary a lot the fan cooling ratio when the model presents small section areas like for example you would do with a hand or a pointy end of an structural part. Yeah, you can modify the print speed but this is not enough a lot of times. I also like to bump up the fan speed at the interface with the supports so they are removed much more easily. This should be very easy to implement, just call that option on the settings window for layer height ratio, modifier elements and specially on separate part settings, which I cannot understand how two different geometries must share the same cooling ratio at this stage! Also it would be nice to have a window on fan speed settings section where you can insert the nr of layer and the % of fan speed you want for that layer, this is because for example I tend to start fan at 60% on the 3rd layer, and increase it to 80% on 5th layer and to 100% on 7th layer since otherwise the deflected air flow from the build plate to the printhead will take too much heat from the hotend.

At this moment is the only thing that I can see simplify3D does great and superslicer/prusaslicer have not even considered the possibility, which is of major importance both for decorative or structural parts (you also want to disable it completely when you need a section which is simple in terms of geometry but needs high Z strenght).

Thanks in advance

neophyl, 2022-04-20:
While not addressing your complete list of wants you can use the insert custom gcode function to insert anything you like at a layer.  Slice it and then use the layer slider to select the layer.  Then instead of left clicking the + icon to insert a colour change use right click and choose insert custom gcode.

pa0los, 2022-04-21:
I second this request. 
Inserting a custom gcode like @neophyl proposed doesn't really help, since the fan speed can be changed (automatically by slicer) multiple times within a layer for different reasons, like overhangs, small features etc.

Hashan0, 2022-04-21:
I also agree with this request.
#2724

Garryb123, 2022-04-22:
Excellent request.

supermerill, 2022-04-22:
duplicate of #478, right?

viniavila, 2025-06-05:
Reviving an old topic, but for documentation:
You can also run code in Printer Settings > Custom G-Code > Before Layer Change, using the parameters `layer_num` (layer number) or `layer_z` (height in mm) to control when to apply the changes. For example:
```
{if layer_num < 269 } M106 S0        ; no fan
{elsif layer_num < 519 } M106 S64  ; 25% fan
{elsif layer_num < 529 } M106 S0   ; no fan
{else} M106 S128 ; 50% fan 
{endif}
```

quickquest88, 2025-11-26:
> Reviving an old topic, but for documentation: You can also run code in Printer Settings > Custom G-Code > Before Layer Change, using the parameters `layer_num` (layer number) or `layer_z` (height in [...]

Yes, but has been said in earlier replies, this is not good enough. The slicer can change fan speeds multiple times over the course of a layer, a single fan speed will never work for an entire layer. This really is needed functionality. 

Example: I am trying to tune bridging on my printer. I have speed, flow rate, and fan speed as my main variables for this. I print four tests at once with the different variables. Otherwise, printing one at a time takes forever. I have a 350, it takes 10 minutes to print, and 45 to cool down to start another print.