supermerill/SuperSlicer #2731: Feature Request: Add a 'Save As' Button

Garryb123, 2022-04-20:
I'd like to see the 'Save' button used just for saving the active profile; one click and done.

I've always wondered why Slic3r, PrusaSlicer, and now SuperSlicer don't have a 'Save As' button for saving settings to a new profile, and instead use the 'Save' button for saving changes to the active profile AND for saving settings to a new profile.

Right now saving a change to the active profile requires three steps: 1) clicking the 'Save' button, 2) moving the mouse to the dialogue box, and 3) clicking ''OK'. This process is much longer - and takes more effort - than necessary. The 'Saving' process could easily be shortened to one click and no subsequent mouse moves if the 'Save' button simply saved the changes to the active profile when clicked.

Saving changes to a new profile would require a 'Save As' button and the steps involved would need to be much like the current process but would make sense for that use.

 Saving to the active profile is probably done 10 times or more often than saving to a new profile and the current method results in a lot of wasted time and effort. This method I suggest is used over and over in the software world and could - I think - easily be adapted to SuperSlicer. I think it would be warmly welcomed by SuperSlicer users and another reason SuperSlicer is better than the other slicers.

Or is there something in the software that would prohibit this functionality?

supermerill, 2022-04-22:
we are talking about adding a new icon next to the floppy disk, a floppy disk with a (+) on it?

Garryb123, 2022-04-22:
> we are talking about adding a new icon next to the floppy disk, a floppy disk with a (+) on it?

For sure! That would be great.

It doesn't take a much different icon; LibreOffice does it like this:

[image]

Thank you for entertaining this idea.

Garryb123, 2022-09-01:
Any idea when you might get to this? You did mark it as 'easy'. :-)

commit 797c469b (2023-11-17): Separate preset save and save-as. Resolves #2731.

This takes advantage of the fact that the `save_preset` function
accepts a name, which hasn't historically been used by the save button.

supermerill:
. redesigned the svg a bit to have a more recognizable "save" icon.
. add set this icon to save project as
. disable "save" when the preset is already saved or isn't modifiable.
. fix a bug with delete icon doesn't enable

supermerill/SuperSlicer#3806
supermerill/SuperSlicer#2731.