supermerill/SuperSlicer #2734: Latest git action 2.4 build produces gcode with crazy extrusion rates

scottmudge, 2022-04-21:
### What happened?

I sliced the same file (attached) with the newest build here:

https://github.com/supermerill/SuperSlicer/actions/runs/2198805654

And a slightly older build here:

https://github.com/supermerill/SuperSlicer/actions/runs/214483692

Same exact settings, same file.

However, when I try to print the gcode from the new version, immediately it's clear that the extrusion rate is wayyy too high for some reason. The extruder starts screaming and the initial skirt line is like 3x thicker than it's supposed to be.

When I print the gcode from the slightly older version, it's fine.

Attached is a zip of the file I sliced, the good gcode (from the slightly older build), and the bad gcode (from the new one).

[example_problem.zip](https://github.com/supermerill/SuperSlicer/files/8534525/example_problem.zip)
2

### Project file & How to reproduce

See above.

### Version

2.4.58.3

### Operating system

Windows 11

### Printer model

SK-Tank

supermerill, 2022-04-22:
I downloaded the first link, and when slicing your project, I have something extremely similar as your good gcode, and very different than your bad one.

you forgot the last '2' in the URL of the second link

Your project file doesn't contain any config information. Please save it via File->save as.

scottmudge, 2022-04-22:
What is wrong with the bad one? Observationally it's clear the extrusion rates are wrong, but I did not go line-by-line with a diff tool to figure out what/where the issue was.

Let me try to re-slice and confirm the problem is repeatable with the new version, and then re-save the session. I did use the "save as" feature, not sure why the config didn't get exported with it.

supermerill, 2022-04-22:
oh, yes, it has config information, my bad.

supermerill, 2022-04-22:
still can't reproduce.

scottmudge, 2022-04-22:
OK thanks for confirming. I'll reinstall the new version and see if I can reproduce. It was reproducible re-slicing during the same software session, but maybe restarting the software will fix it. Perhaps it entered some weird undefined state.

scottmudge, 2022-04-22:
Strange, using the latest (previously problematic) build I resliced the original file and the result was fine.

I wonder if some setting or variable got transiently stuck somehow. I did not see any modified settings or parameters asking to be saved when I closed the software, though. 

Or perhaps my RAM is going bad or got hit with a cosmic ray... who knows. 

Thanks for double checking. Closing.

closed 2022-04-22 by scottmudge