supermerill/SuperSlicer #2735: Crash on slicing when selecting scattered rectilinear infill

nwpulver, 2022-04-21:
### Version
Version 2.3.57

### Operating system type + version
Linux: Manjaro KDE

### Behavior
Turning on scattered rectilinear infill for the sparse infill causes super slicer to crash once press slice. 

#### Project File (.3MF) where problem occurs
File is not zipper, i just renamed it zipped. 

[bug.zip](https://github.com/supermerill/SuperSlicer/files/8535165/bug.zip)

supermerill, 2022-04-22:
when opening your project file, it was on adaptative cubic and not scattered rectilinear, is it a bug from the slicer?
when slicing it with both adaptative cubic and scattered rectilinear on the 2.3.57.12, it doesn't crash.
Are you sure you are on the 2.3.57.12 version?

nwpulver, 2022-04-27:
Directly from system info: Version: 2.3.57.12
Build: SuperSlicer_2.3.57.12_2022-03-31

Here is a video: Again not actually zipped file - This is a .mp4 file from OBS. It happens every time for me. 
[2022-04-26_19-42-13.zip](https://github.com/supermerill/SuperSlicer/files/8568314/2022-04-26_19-42-13.zip)

supermerill, 2022-04-28:
My computer say: 2022-04-26_19-42-13.zip is not a zip archive

nwpulver, 2022-04-29:
rename it to *whatever name*.mp4 

It is not a zip file- i just renamed so I could upload here.

Miwer, 2023-08-07:
Hi,

I have the same crash, when selecting "Scattered Rectilinear" infill.

Also Manjaro KDE user here.
Although I'm using the latest SuperSlicer version 2.4.58.5 installed via this AUR package:
https://aur.archlinux.org/packages/superslicer-bin
([SuperSlicer-ubuntu_18.04-2.4.58.5.AppImage](https://github.com/supermerill/SuperSlicer/releases/download/2.4.58.5/SuperSlicer-ubuntu_18.04-2.4.58.5.AppImage))

It seems when the crash occurs, there's a general protection fault in libc.so.6, and the core is dumped.

I'm attaching a trace log from SuperSlicer (log ends at the crash with no details), and the associated journal syslog showing the core dump.

Hope it helps to point toward identifying a cause and a solution.
Thanks.

[superslicer_journal.log](https://github.com/supermerill/SuperSlicer/files/12278886/superslicer_journal.log)
[superslicer_trace.log](https://github.com/supermerill/SuperSlicer/files/12278887/superslicer_trace.log)