supermerill/SuperSlicer #2736: Regression: "Avoid Crossing Perimeters" constantly being violated, wipe moves also crossing perimeters

thwerks, 2022-04-22:
### What happened?

Avoid Crossing Perimeters setting is violated constantly. there seems to be no explanation for this. 

Last known version without this issue was 2.3.56.9. Issue starts creeping in to the scene by version 2.3.57.11 and progressively getting worse. Now with  2.4.58.2 is completely just mangled, virtually no difference if its enabled or not. 

Now in 2.4.58.2 even wipes are crossing the perimeters.  Thin Wall option slightly improves problem, sadly not nearly enough.

[image]

### Project file & How to reproduce

[Insert 1-2.txt](https://github.com/supermerill/SuperSlicer/files/8537005/Insert.1-2.txt)

rename file to  .3mf

### Version

2.3.56.9, 2.3.57.11 and 2.4.58.2

### Operating system

Debian 11

### Printer model

Ender 3 Pro

m-sundman, 2022-04-27:
This problem has been bothering every single slicing I've done with SuperSlicer in a very long time, and is driving me crazy. What is needed to get this fixed? How can I help?

supermerill, 2022-04-28:
> What is needed to get this fixed? How can I help?

probably changes made to improve the feature (to not follow the first perimeter but a bit more inside) has these undesirable side-effects. To solve that, I (or someone else) have to create a minimal one-layer object that reproduce the bad behavior. Then export the different polygons & paths at many steps (in the AvoidCrossingPerimeter.cpp source code). With these input, I should be able to identify where the code is failing to fall back to the degraded path (not on the second perimeter as there isn't one), and fix it.

MetalFranke, 2022-08-04:
Hello,

I digged a little into the "wipe is crossing perimeters" problem and found out that it happens at the inside of holes only (with 2.4.58.3). The wipe on the top of the picture seems to be correct the one in the hole is not:

[image]

It seems GCode.cpp, line 4035 is wrong. Since holes are extruded clockwise the "dist" is still positive there.
`Polygons polys = offset(original_polygon, (original_polygon.is_clockwise()?1:-1) * dist);`

Taking the current master code and changing to a negative "dist" (?-1:1) I get this imho correct result. It may be discussed if the little wipe inwards at the beginning should be limited to the 50% nozzle diameter also.

[image]

supermerill, 2022-08-04:
@MetalFranke thank for the debug.
That is_clockwise check shouldn't be done, as the offset takes care of that.
I'll merge the fix.

nathanshearer, 2023-10-30:
This is impacting me on 2.5.59. I have been struggling to get prints without blobs and it turns out it has been wiping outside my part, but only on the inside surface of holes. The "Wipe Inside" feature is actually causing blobs that would otherwise be prevented by back-pressure during a typical wipe over already material and is also impacting dimensional accuracy on the outside surfaces.

As a temporary workaround I have set:
- Wipe while retracting (wipe) to enabled
- Minimum travel after retraction (retract_before_travel) to 0.8 to force retraction when moving from an outside perimeter to another perimeter or infill with a 0.4mm nozzle and 3 or more perimeters
- Wipe only when crossing perimeters (only_retract_when_crossing_perimeters) to disabled
- Wipe inside at start and end (wipe_inside_start and wipe_inside_end) to both disabled

This causes the nozzle to retract at the end of every perimeter while retracting, for the length of the retraction that is specified in "Retraction Length" (retract_length). Filaments or configurations that require more retraction will wipe for longer as needed.

Hopefully when this is fixed and merged into the newest version I can try enabling the "Wipe Inside" feature again.