supermerill/SuperSlicer #2737: Wrong G-Code being injected...

dubiyus, 2022-04-22:
### Version
; generated by SuperSlicer 2.3.57 on 2022-04-22 at 04:16:24 UTC

### Operating system type + version
Win 10 Pro 21H2

### Behavior
Generated G-Code is just wrong!
Slice anything....
Extruder is set to extruder temp
Extruder is set to bed temp

Fix your gcode.
[image]

neophyl, 2022-04-22:
Its strange that it works for mostly everyone else.  So that would suggest that its something in your configuration/setup that is awry. 

Without a saved project file (zipped up 3mf as requested by the issue template when opening an issue) then its going to be impossible to figure out why you specifically are having a problem.  With a project file then the developer (or anyone else interested) can check for any misconfiguration as the project will contain a copy of all the settings and what is configured on the profiles you have selected at the time of saving the project.

dubiyus, 2022-04-23:
Sorry about that! Here you go:
[issues.zip](https://github.com/supermerill/SuperSlicer/files/8545948/issues.zip)

closed 2022-04-23 by dubiyus