supermerill/SuperSlicer #2740: Object manipulation not visible

Mrugacz, 2022-04-22:
### What happened?

Object manipulation doesn't show any options.
[image]

### Project file & How to reproduce

Import file -> object manipulation section is empty

Problem appears while using tiling window manager, in a proper desktop environment (gnome) it works as intended

### Version

2.4.58.0

### Operating system

Linux 5.17.4-zen1-1-zen x86_64

### Printer model

unrelated

Gusser93, 2022-09-21:
~Not working at my gnome environment with wayland.~ Not working with some dark themes. Works when starting with `GTK_THEME=Adwaita:light superslicer` and `GTK_THEME=Adwaita:dark superslicer`. I can manipulate the fields but I can not see them since using this tool last year. Missed this feature until I saw that it should have been there…
Here my not working config:
| | |
| -- | -- |
| Version: | 2.4.58.4 |
| OS: | Arch |
| GL version | 4.6 (Compatibility Profile) Mesa 22.1.7 |
| GLSL version | 4.60 |
| DE: | GNOME 42.4 |
| WM: | Mutter |
| WM Theme: | Materia-dark-compact |
| Theme: | Materia-dark-compact [GTK2/3] |

ronasimi, 2024-02-19:
I have the same issue on Arch in hyprland, using Materia-dark-compact

supermerill, 2024-05-21:
found some `SetForegroundColor(BLACK)` in the class that render this, while in evrywhere else it's `SetForegroundColor(get_label_color())`, so I changed it.
I don't have a dark theme, but I guess this will fix it.
Tell me if not (will be in next nightly)

closed 2024-06-12 by supermerill

supermerill, 2024-06-16:
now it seems broken for windows, as the background is white...