supermerill/SuperSlicer #2741: wipe tower retraction, dual extruder, single nozzle

phongshader, 2022-04-23:
### Discussed in https://github.com/supermerill/SuperSlicer/discussions/2730

<div type='discussions-op-text'>

<sup>Originally posted by **phongshader** April 20, 2022</sup>
 What is happening is, with NO filament swap, the current filament retracts to park position when transitioning from part to wipe tower. This seems to be a huge waste of time and also causes filament jams in my case. Is this by design? Wouldn't a simple retraction suffice in this situation? Is there a way to get the settings so when there is NO filament change a simple retraction, ~2-5mm,  is used instead of retracting to the park position, ~95mm, when transitioning to the wipe tower? </div>

phongshader, 2022-04-24:
I can post process the gcode to remove and replace all the E95 and E-95 moves with E5 and E-5 and the print time drops considerably...this should be able to be handled by the slicer...no?

supermerill, 2022-04-28:
why using "single extruder mmu" if you have a dual extruder?

supermerill, 2022-04-28:
why do you have a "Single extruder mm setup" section in your screenshot?

phongshader, 2022-04-28:
**_FACE PALM_** you are right...I apologize for not seeing that...it all makes sense now...
I was focused on single nozzle not single extruder.

phongshader, 2022-04-29:
so even with that changed I still have the same issue...too much retraction. What is the correct way to set up dual extruder SINGLE nozzle? If I uncheck Single Extruder Multi Material I lose any configuration settings for filament park, cooling tube position and so on. I still need those to tell ss where to retract to when the filament changes....which is odd because ss still retracted 95mm with each layer change when purging on the wipe tower even though Single Extruder Multi Material was disabled.

supermerill, 2022-04-29:
Oh sorry, if you have a single nozzle, then you may really want the "Single extruder mm setup".
I don't really know how to tune it, I don't use it much. I guess it's possible to do what you want.

The last time I had to use a kind of mmu with one nozzle, I wrote the gcode swap code directly in the toolchange custom gcode (as I wanted to do unorthodox stuff).

I'll try to convert my mmu thing to the "Single extruder mm setup", to be able to answer you.

phongshader, 2022-04-29:
from my other thread:
OK, lightbulb goes off in head...use Single Extruder Multi Material but set Minimum retraction to 0. So the way to set up dual extruder single nozzle is to SELECT Single Extruder Multi Material in Printer Settings > General, then Printer Settings > Single Extruder MM Setup set the correct values for Cooling tube position, cooling tube length, and filament parking, then Printer Settings > Extruder1 and Extruder2 set minimum retraction to 0. This is working for me. THANK YOU for all your help! This should be added to some documentation somewhere.