supermerill/SuperSlicer #2743: M600 and M601 with Klipper

XoBoR, 2022-04-23:
When inserting a Pause on a certain layer with G-Code flavor Klipper selected, M601 is used, which is not supported by Klipper, the same is true for M601.

**Expected behavior**
Insert a [PAUSE](https://github.com/Klipper3d/klipper/blob/master/docs/Config_Reference.md#pause_resume) instead.

**Additional context**
Currently you can work around the problem by either using G-Code Substitution to change M601 into PAUSE or defining a Macro named M601 in klipper or of course inserting custom G-Code.

neophyl, 2022-04-23:
There are fields under printer settings where you can define your own gcode for those functions. It’s where the m600 and m601 are defined. Simply change them.

schnello, 2022-04-24:
Issue solved ;)
[image]

XoBoR, 2022-04-24:
Oh, i totally missed that, sorry ;)

closed 2022-04-24 by XoBoR

your-friend-alice, 2023-10-01:
uh, that's not the solution to the issue, that's a workaround. what's the point of defining a "klipper" preset if its output is incompatible with klipper?

schnello, 2023-10-01:
Create a M601 Macro.. solved again

your-friend-alice, 2023-10-01:
are you opposed to the idea of things working correctly by default? i do not understand this attitude, each incompatibility adds up and makes the experience for new users worse for no good reason. I'm going to make a PR now, is that going to be a problem?

schnello, 2023-10-01:
Iam working with Klipper now a long long time and ist works. So if you are the opinion this should be a default feature ... yes please make a PR. Better as posting in old closed topics...

your-friend-alice, 2023-10-02:
my apologies, I got a bit more heated than necessary there. I tend to bristle when people conflate "workarounds" and "solutions" (as I believe it gets in the way of building more usable software in the long term), but that's my own pet peeve to bear. You were literally helping someone get their setup working, that's actually good and cool. PR is up to hopefully save some future people the trouble https://github.com/supermerill/SuperSlicer/pull/3943.