supermerill/SuperSlicer #2745: No Z lift at first layer in some rapid movements

AMT3D, 2022-04-24:
Hello!!

### Version
Superslicer 2.4

### Windows 10 64Bit

### No Z lift 
* Does anyone knows why at first layer some rapid movements doesn't lift the amount required. Is not and all movements I believe is mainly the case when filling gaps from one side to another side.
* Even using up to 5mm Z lift will not improve.
* This leave some marks in the first layer that in my case is the cosmetic view of the part. Parts are coming with good quality no mayor issue but i wonder why not do the Z Lift 
* Z lift 0.5mm
* Avoid crossing perimeters "ON"
* Not in first layer "Not On"
 Changing any combination of this settings doesn't change the outcome.
In the picture below you can see the rapid movement not lifting creating a small string of material.
Direct Drive with 2mm retraction @30mm/s 
[image]

neophyl, 2022-04-24:
You haven’t included the requested zipped up project when opening an issue. There are so many interactive settings that a project is required to debug most issues.

AMT3D, 2022-04-24:
[Frame.zip](https://github.com/supermerill/SuperSlicer/files/8549561/Frame.zip)
Hello 
Sorry for that. 
Attached is the superslicer 3mf and the original STL

supermerill, 2022-09-07:
seems fixed now.
reopen this if not.

closed 2024-02-11 by supermerill