supermerill/SuperSlicer #2748: Extreme Infill/Perimeter Knitting [Feature Request]

Garryb123, 2022-04-24:
I'm wondering if you could program infill/perimeter knitting to extreme levels as in this picture? This would be useful for larger parts that can use more perimeters and need maximum strength.

Being able to choose 1) how far the infill extends into perimeters, 2) how many perimeters the first layer has, and 3) even being able to specify alternating infill distances (bottom left image) into perimeters would be epic.

I don't know how difficult this would be to program so maybe it's too much. But maybe it's doable.

[image]

supermerill, 2022-04-28:
currently, the first one is possible with extra perimeters -> on odd.

>  This would be useful for larger parts that can use more perimeters and need maximum strength.

do you have data for that?
Currently, you can simulate that with modifiers (not user-friendly, but good enough for a prototype). If you can prove that it increases the strength, I'll modify the GUI so you can have that.

Garryb123, 2022-04-28:
Thank you; yes I'm using the extra perimeters on odd. It's good.

Sorry, I don't have any data to back up my idea.  Also, I don't know how to go about using modifiers to test it. I'm not an engineer so have ideas and then put them into practice and see how they turn out ('Trial and error'). My hunch is this should be useful for increasing infill/wall bonding but I have no scientific proof.

supermerill, 2022-04-29:
Enable the "height range modifier" on your object
set the stop at height at the layer height value
then add more with the (+), and set the new "stop  at height"
repeat until your object is fully covered.
then, add the perimeter count setting to the first range.
copy-paste it to all others.
then, you can modify the perimeter count to simulate your knitting.

Garryb123, 2022-04-29:
Thank you for explaining the process. I will test this and report back.

Garryb123, 2022-04-30:
Again, thank you again for the detailed instructions; they worked perfectly and the effect is as desired. 

I'm disabled so can't perform a test to break the part to see how strong it compared to a current part but believe infill/loops done this way MUST be stronger due to the amount of infill (sometimes solid) that is enmeshed with the loops.

I thought you would need to change the infill/loop algorithm to achieve this effect and imagined that could be a lot of work - maybe impossible - but if you just need to change the GUI I hope you can do this.

All the best.

supermerill, 2022-05-02:
> I thought you would need to change the infill/loop algorithm to achieve this effect and imagined that could be a lot of work - maybe impossible - but if you just need to change the GUI I hope you can do this.

It's not impossible, not even that hard. But why add it if there is no need for it? That's why I ask for a real use-case. Also I have to defend the feature if someone ask me "why did you add that".

The 'add perimeter on odd layer' is here to sandwich & support the infill to be able to use a low /no perimeter/infill encroachment. There is no "strength" purpose.

Garryb123, 2022-05-02:
Thanks. I always thought that option was for binding the infill and perimeter and my idea was an expansion on that. If the setting isn’t for strength then I guess my theory isn’t valid.