supermerill/SuperSlicer #2749: Some print settings drive vase mode nuts and generate travels

sprior, 2022-04-24:
### What happened?

I printed a large vase mode object with a 1.2mm nozzle and 0.8 layer height and found severe problems with the created gcode.  It generated travel moves through the print and the perimeters are also distorted in the gcode preview.  I have reproduced this in both 2.3.57.12 and 2.4.58.3 (nightly from a few days ago).  I then found that if I created a new print profile from a slightly different source I appear to generate ok gcode, but while I compared the settings for both I don't see anything which should cause such issues.

Attached are bad.jpg which shows the wonky gcode layers as well as the travel moves which shouldn't be there as well as good.jsp which appears fine.  Also bad.init.txt and good.init.txt which shows all the differences between the two files.  I'll also attach the original model stretchlet1.stl and the bad project saved from that with the bad settings.

![bad](https://user-images.githubusercontent.com/33618
[image]
00/164997790-950c9e8e-3877-4212-b6c4-7e079eca9628.jpg)
[good.ini.txt](https://github.com/supermerill/SuperSlicer/files/8550552/good.ini.txt)

[bad.ini.txt](https://github.com/supermerill/SuperSlicer/files/8550551/bad.ini.txt)

### Project file & How to reproduce

[project.zip](https://github.com/supermerill/SuperSlicer/files/8550549/project.zip)

### Version

both 2.3.57.12 and 2.4.58.3 (nightly from a few days ago)

### Operating system

Windows 10

### Printer model

Voron 2.4 with LGX

sprior, 2022-04-24:
[updated_project.zip](https://github.com/supermerill/SuperSlicer/files/8550568/updated_project.zip)

Here's an updated project.zip which contains the full print settings for both bad and good.  Since Super Slicer seems to get confused enough to mangle the generated 3mf file you may have to recreate from the original stl which I've included.

supermerill, 2022-04-28:
The problematic parameter is `gap_fill_last` (deactivate it)
I'll fix

closed 2024-02-11 by supermerill