supermerill/SuperSlicer #2756: Calibration  > extrudeur calibration

Nexpyro, 2022-04-26:
### What happened?

Tower is not generated with correct pitch if you choose it from the drop down menu.
the drop down menu as 0.1 / 0.2 / 0.5 / 1 / 2. If clicking on 0.2 or 0.5 the tower is generated with the default 0.1 pitch

To make it work we need to have 0,1 / 0,2 / 0,5 / 1 / 2 in the slicer is in french.

### Project file & How to reproduce

Superslicer in french . maybe it's related to some languages that use "," instead of "." because everything work when i come back to english.
May affect other think in slicer because i'ts maybe a general bug for french language

### Version

2.4.58.2

### Operating system

Win 10

### Printer model

ender 5

supermerill, 2022-04-26:
in the 2.4, prusa has the ?good? idea to respect the ',' vs '.' for separating the int part and the flat part of a real number.
I have to ~loose my time to~ modify all my additional GUI items, changing their behaviour to fix these kinds of issues.

RedSamura, 2022-12-13:
This problem is relevant at the moment and is observed in most European languages. In English and Japanese - works correctly.