supermerill/SuperSlicer #2757: usually nighlty macos are around 57mb, sometime around 37 and they don't open

wnc2, 2022-04-26:
the issue happens often.

last (not working) version:
[image]

last but one (working) version:
[image]

supermerill, 2022-04-28:
yes. The macos build has a problem: when it fails, it doesn't show itself as failed but and publish a dmg without the software inside.

wnc2, 2022-04-30:
latest is still bugged, why you publish them if they are bugged?
[image]

supermerill, 2022-04-30:
> latest is still bugged, why you publish them if they are bugged?

The macos build has a problem: when it fails, it doesn't show itself as failed but and publish a dmg without the software inside.
I don't publish these, it's just the output of the automated builds.
What I publish is in the releases page.