supermerill/SuperSlicer #2761: Horrible "internal bridge infill" quality

LaserBee, 2022-04-26:
### What happened?

Okay, so I believe it's the "Internal Bridge Infill" that is giving the issue, Basically it's barely attached, leaves loads of little bumps, bits of stringy bits, &c. 

Here is how it looks in the slicer:

[image]

And here is how it looks when it's printed (note I cancelled the print when it happened!)

[image]

The next layer (or the one after) normally ploughs through the rough bits and the final layer looks alright, but I'm paranoid that it will eventually cause a failure when the nozzle strikes a bump. 

Any idea what is causing this? 

 Attached find the project file...

### Project file & How to reproduce

[Wheeliebase.zip](https://github.com/supermerill/SuperSlicer/files/8566908/Wheeliebase.zip)

supermerill, 2022-04-26:
it's the denser supporting layer, you can deactivate it in the infill section.
It's to ensure that any solid infill has something to lay onto, as the sparse infill can be too sparse.
https://github.com/supermerill/SuperSlicer/wiki/Dense-infill-under-solid-layer

LaserBee, 2022-04-27:
Thank you, Supermerill. I realise I've got my "INTERNAL BRIDGES" set at 150%, so I will try lowering that to 80% and see if it makes a difference.

slavedas, 2022-05-28:
I ran into the Internal Bridges speed issue too.  I THINK it's supposed to be 150% of the Bridges speed, but it's actually 150% of the default speed.

supermerill, 2022-07-12:
> I THINK it's supposed to be 150% of the Bridges speed, but it's actually 150% of the default speed.

can't reproduce that, do you have a project that show that behavior?

Tikonderoga, 2024-01-24:
@LaserBee 
Did you figure it out? I've stumbled upon the same issue.