supermerill/SuperSlicer #2762: Setting for Overhang Angle

wuast94, 2022-04-27:
**Is your feature request related to a problem? Please describe.**
I print really fast and Overhangs don't get cooled enough, i can set an overhang speed override but i cant set the angle that is recognized as an overhang. 

**Describe the solution you'd like**
A Setting where i can set an angle that specifies at which point the overhang speed override should be used.  

**Describe how it would work**
I think the Images make it more clear, the blue part at the bottom is recognized as overhang, i print around 180mm/s. Looks fine on walls etc. but the overhang looks horrible, no wonder.
The only thing i can do atm is to slow down my printer on all walls, but its only necessary to a specific overhang angle.
if i could set this angle i could maintain much more quality without slow down the whole print. 

[image]
[image]

Hope i don't missed any settings for this, and my englisch isn't the best so i hope you understand what my problem and suggestion is 🙂

neophyl, 2022-04-27:
Cooling issues in that area on the benchy aren't generally caused by the overhang directly, its usually the slicing settings adding extra solid infill next to the perimeter walls due to the slope.  This extra plastic means much more heat in that area.

If you had attached the requested zipped project file when opening the issue we could slice with your settings and examine the preview ourselves to see if your current settings are adding that extra infill or not.  

Under Perimeters & Shell > Overhangs there are the thresholds for Bridge speed & Fan and also bridge flow.  You can tweak those to determine what 'angle' SuSi decides a perimeter is an overhang.  That then kicks in the cooling and/or flow assigned for bridges/overhangs.  One thing though, its not an angle that is the setting but is based on how much the perimeter overlap the one below.  In effect it allows you to do the same thing though.

wuast94, 2022-04-27:
Sry missed that point, here it is 🙂
[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/8570458/3DBenchy.zip)

I understand that the heat can come from other places, but having more  controll about overhang speeds would be really helpful anyway i think.

supermerill, 2022-04-28:
> I understand that the heat can come from other places, but having more controll about overhang speeds would be really helpful anyway i think.

neophyl have already answer where you can modify these
> Under Perimeters & Shell > Overhangs there are the thresholds for Bridge speed & Fan and also bridge flow. You can tweak those to determine what 'angle' SuSi decides a perimeter is an overhang. That [...]

It's for the overhang detection threshold.
TO control the overhang speed, it's under speed tab -> spedd->bridge -> overhangs

supermerill, 2022-04-28:
It's a bit slow (or difficult) to compute.

wuast94, 2022-04-28:
in normal cases yes, i print with an accel from 4000mm/s². so i print at max speed everytime almost. the metric that limits me at the moment is the flow rate.

i get the front fixed now, the problem is that the filament looks very diffrent at the overhangs, i have a matt finish and on this areas its glossy now. i need to play more with the lower speeds to sort this out i think. at low speeds i get ringing too. with a dynamic calculation it would be a more fluent change and not this hard change. but i will see if i get my overhang % and speed tuned right. i hope for a sweet spot beetween this too problems

wuast94, 2022-04-28:
190 and 40, i know the diffrence is way to much, was just a fix test to see if it works, i need to find the perfect values now