supermerill/SuperSlicer #2765: Internal Bridge Infill Selective Perimeter

second-pair, 2022-04-28:
#  Feature Request:  Internal Bridge Infill Selective Perimeter
Version:  SuperSlicer 2.3.57.12
[internal-bridge-infill-mre.zip](https://github.com/supermerill/SuperSlicer/files/8582159/internal-bridge-infill-mre.zip)

##  Background Info
_Internal Bridge Infill_ often prints onto thin air, as mentioned in [Issue #1185](https://github.com/supermerill/SuperSlicer/issues/1185#issuecomment-858220792).  A number of solutions were proposed:
1.  Using _Supporting Dense Layer_ is undesirable, as this uses substantially more filament, where a single perimeter line to anchor the _Internal Bridge Infill_ to would suffice.  In my mind, this setting has its own place, where this layer needs to be printed with a high quality.
2.  Setting __infill_anchor_max__ and __infill_anchor__ to high values (such as 10000) doesn't solve this, as the _Internal Bridge Infill_ is often printed before the _Internal Infill_'s perimeter.  This would typically use more filament than Solution 1 anyway, unless I were to painstakingly apply this to only the relevant layers.
3.  This would use much more filament than Solution 1 and wouldn't completely solve the issue if using __fill_pattern__ == Gyroid.

##  Demonstration
The attached project is, I believe, the Minimum Reproducible Example for this behaviour and will consistently demonstrate this behaviour:
####  3D View
[image]
####  Bridge Infill
[image]
####  Large 'infill_anchor'
[image]
####  Infill Perimeter Printed Second
[image]

##  The Solution I'd Like
I would like to request a setting to selectively print perimeter lines for the _Internal Bridge Infill_ to anchor to.  I believe this would solve my issue while requiring the minimal amount of filament to do so.  It would make sense to me for such lines to extend beyond the edge of any _Internal Bridge Infill_, to ensure it is itself supported correctly - I've tried to demonstrate this in this Paint mock-up:
####  Solution Mock-Up
[image]

Thanks!

supermerill, 2022-04-28:
The solution mock-up is already here, as this is the rectilinear-with-perimeter infill pattern (used for supporting dense infill). But currently, bridge infill is forced to use the rectilinear.

I can add an option to change the pattern of the internal bridge infill pattern to rectiWithPeri. Is it enough?

second-pair, 2022-04-28:
Yes, I think that would be perfect!