supermerill/SuperSlicer #2770: Setting Sets as toggleable features

piotrb, 2022-04-29:
**Is your feature request related to a problem? Please describe.**
One problem I tend to run into a lot is that I end up having to manually "sync" various sets of settings into print profiles .. 
They are all the same values .. like for example:

* brim enable + size
* skirt enable & number number of skirt loops
* or infill pattern and density
* or settings specific to nozzle size 

**Describe the solution you'd like**
I'd love to be able to layer small subsets of settings in-between the filament level overrides and the print settings .. 

It'd have to be a list .. or maybe check boxes .. and you can drag them around to set the order, and enable/disable them as needed.

**Describe how it would work**
There is already some kind of a settings cascade in place .. filament > print (and printer somewhere in that chain too) .. this would inject somewhere in there .. 

**Describe alternatives you've considered**
There isn't good ones .. have to manually copy settings between print profiles right now .. 

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2022-04-29:
if .. maybe it's ... but i'm not sure ...

piotrb, 2022-05-02:
I guess the main problem is just switching between profiles .. there is too much tied to this idea of a "profile" .. 
If layer height is the main differentiation between profiles .. then stuff like how I like to my infills .. is kind shared between profiles .. I'd love for that to managed a bit separately so then you can "layer" in those settings .. if this helps ..

supermerill, 2022-05-02:
Not possible, unless the entire profile concept is reworked.

piotrb, 2022-05-02:
Hmm perhaps .. an easier place to inject this may be at the per-part setting overrides? Make it easier to add specific pre-configured sets of options to a part?

supermerill, 2022-05-02:
You can do that with modifiers.
right-click -> settings

piotrb, 2022-05-02:
Right, but you configure them fresh each time right? I have not seen an option to create modifier presets .. even that UI that pops up asking which individual settings to add .. it may be nice to be able to save those into a preset .. 

kind of like how right now there is the 3 presets there `Perimeters Shell`, `Infill` and `Support Material` .. it'd be nice to add custom presets like that there .. since selecting options from that "custom" ui .. is a fair bit of a pain .. its a flat list of settings, often I'm unsure if I'm selecting the right one (adding the setting descriptions there in a tooltip may be a nice help) .. but also creating presets of "selected override settings" may be really nice ..

supermerill, 2022-05-02:
that is feasible.

neophyl, 2022-05-03:
Piotrb - one thing some of us do (with both PS and Susi) is to use small objects as 'libraries'  .  You create a small non printing object.  Then you can create one or more modifiers assigned to that object.  These objects become the repository, that extra profile if you will.  You can then use cntrl+c/cntrl+v to copy and paste a modifier from one object to another.  

You then save the project.  That project becomes your template.  Instead of using New project you just load a copy of the saved project.  That way you get your favourite pre-sets selected anyway and all the modifiers you have created on your library objects which can then be applied to the models you load.  
As you can save as many projects as you want this means you can store different templates for different printer combo's or different slicing scenarios.  Its open ended so up to you.

Its on 'off books' way of using the current project system to your advantage.