supermerill/SuperSlicer #2771: Add Part no longer works as in 2.3.x

marc-gist, 2022-04-30:
### What happened?

When I right click and do "add part" under 2.4.58.2, it does not slice with the model, only parts that are above the original model slice.

### Project file & How to reproduce

[add_part_slicer_not_working_2.4.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8595516/add_part_slicer_not_working_2.4.3mf.zip)

do the same thing in version 2.3.5xxx and it slices as expected (or how I used it in the past :) ).

### Version

2.4.58.2

### Operating system

Windows 10

### Printer model

Voron, but doesn't matter.

supermerill, 2022-04-30:
Can't reproduce.
[image]
what I have to do? and what is the bad result?

Henyckma-Sensei, 2022-06-02:
hi i have the same issue, sometimes it works and sometimes it doesn't, it also depends on the parts and sometimes it only loads half of the part that doesn't intersect with the model