supermerill/SuperSlicer #2772: Lift Z at top layer not applied properly with Infill before perimeters

Dilbao, 2022-04-30:
### What happened?

I want to use z lift on my prints, without z lift movements leave visible mark on the surface.

### Project file & How to reproduce

Here is a summary my results with the test cube:

**Lift Z enforcement: All surfaces**
Lift Z every where, every layer. This is works but too many unnecessary lifts.

**Lift Z enforcement: Only on top & Infill before perimeters: false**
Perimeters extruded without z lift then top infill extruded with z lift. This is works but order is wrong.

**Lift Z enforcement: Only on top & Infill before perimeters: false**
Top infill extruded with z lift then perimeters extruded without z lift and this leaves marks on the top surface.

### Version

v2.4.58.3 (nightly@0430)

### Operating system

Windows 10

### Printer model

Ender 3 v2

supermerill, 2022-04-30:
and what do you want? to also lift for all perimeters?

Dilbao, 2022-04-30:
Expected behavior for "z lift over top surfaces" should be like not traveling over any top surface without z lift. Default behavior is not sufficient for this.

If you print perimeters first (which is not lifting between moves), infills cover any defect later. But if you print infills first, movement between perimeters will leave marks over previously filled surfaces.

supermerill, 2022-05-02:
did you set your software in a different language as english? if yes, can you try with english?

Dilbao, 2022-05-02:
I deleted profiles folder, created new printers with default settings.
[Uploaded the project file.](https://github.com/supermerill/SuperSlicer/files/8605606/Default.Settings.-.Cube.-.No.Lift.zip) Only changes are:

> Printer Settings > Extruder 1 > Retraction > Lift Z = 0.2
> Printer Settings > Extruder 1 > Retraction > Only lift Z > On surfaces = Only on top
> Print Settings > Infill > Advanced > Infill before perimiters = true

After slicing, check line 31880. Letter Z finished and next is outer perimiter. But there is no lift between them.