supermerill/SuperSlicer #2773: Z-Hop leads to print issues/empty layers in latest nightly

codingcatgirl, 2022-05-01:
1. Add a cube to the print surface
2. Select Default Printer: Original Prusa i3 MK3S MMU2
3. Select Default Preset: 0.20 QUALITY
4. Select Any Filament

Result: 
[image]

Disabling Z-Hop in Extruders and Filaments fixes the issue.

codingcatgirl, 2022-05-01:
After some further investigation i can confirm that this is not just a rendering error. 

---

Layer change to layer 1.0mm shows the correct behavior (relevant lines highlighted)

> _**;LAYER_CHANGE
> ;Z:1**_
> ;HEIGHT:0.2
> ; custom gcode: before_layer_gcode
> ;BEFORE_LAYER_CHANGE
> G92 E0.0
> ;1
> 
> ; custom gcode end: before_layer_gcode
> 
> ;WIPE_START
> G1 F8640
> G1 X121.196 Y100.929 E-0.14416 ; wipe and retract
> G1 X121.771 Y100.929 E-0.13293 ; wipe and retract
> M73 Q34 S3
> G1 X120.929 Y101.771 E-0.27499 ; wipe and retract
> M73 P34 R3
> G1 X120.929 Y102.347 E-0.13293 ; wipe and retract
> G1 X121.159 Y102.117 E-0.07498 ; wipe and retract
> ;WIPE_END
> _**G1 E-0.04 F2100 ; retract
> G1 Z1.4 F10800 ; lift Z**_
> ; stop printing object Shape-Box id:0 copy 0
> M486 S-1
> ; custom gcode: layer_gcode
> ;AFTER_LAYER_CHANGE
> ;1
> ; custom gcode end: layer_gcode
> 
> ; INIT printing object Shape-Box id:0 copy 0
> **_G1 Z1.6 ; lift Z_**
> ; printing object Shape-Box id:0 copy 0
> M486 S0
> M204 S800 ; adjust acceleration
> G1 X120.632 Y100.632 ; move to first Internal perimeter point (minimum acceleration)
> _**G1 Z1 ; restore layer Z
> G1 E0.8 F2100 ; unretract**_
> ;TYPE:Internal perimeter
> ;WIDTH:0.45
> G1 F900
> G1 X129.368 Y100.632 E0.2957 ; Internal perimeter

Z is lifted after retraction and restored before unretracting and before starting extruding the perimeters. However, the z lift is performed twice - which doesn't cause an issues because absolute Z coordinates but seems related to the issue?

---

Here's the layer change to layer 1.2mm, which shows the incorrect behavior:

> **_;LAYER_CHANGE
> ;Z:1.2_**
> ;HEIGHT:0.2
> ; custom gcode: before_layer_gcode
> ;BEFORE_LAYER_CHANGE
> G92 E0.0
> ;1.2
> 
> ; custom gcode end: before_layer_gcode
> 
> ;WIPE_START
> G1 F8640
> G1 X127.184 Y108.934 E-0.07133 ; wipe and retract
> G1 X127.621 Y108.946 E-0.10089 ; wipe and retract
> G1 X128.163 Y109.063 E-0.12818 ; wipe and retract
> G1 X129.063 Y109.063 E-0.20765 ; wipe and retract
> G1 X129.063 Y107.971 E-0.25194 ; wipe and retract
> ;WIPE_END
> **_G1 E-0.04 F2100 ; retract
> G1 Z1.6 F10800 ; lift Z_**
> ; stop printing object Shape-Box id:0 copy 0
> M486 S-1
> ; custom gcode: layer_gcode
> ;AFTER_LAYER_CHANGE
> ;1.2
> ; custom gcode end: layer_gcode
> 
> ; INIT printing object Shape-Box id:0 copy 0
> ; printing object Shape-Box id:0 copy 0
> M486 S0
> M204 S800 ; adjust acceleration
> G1 X129.368 Y109.368 ; move to first Internal perimeter point (minimum acceleration)
> **_G1 E0.8 F2100 ; unretract_**
> ;TYPE:Internal perimeter
> G1 F900
> G1 X120.632 Y109.368 E0.2957 ; Internal perimeter

Z is lifted after retraction but **not** restored before unretraction and continuing extrusion. The Z lift is also only performed once.

codingcatgirl, 2022-05-01:
If i disable the "retract on layer change" setting for all extruders, the problem persist. But in this case:
- during the 1mm layer change, no retracts and wipes are performed during the 1mm layer change at all (sounds right to me)
- during the 1.2mm layer change, the retract and lift is still performedthe retract and the unretract is missing the z restore

supermerill, 2022-05-01:
Can't reproduce. The cube print fine. Using the `fixes, more preferences options, slowing gapfill if needed. C/C++ Nightly windows x64 #578` nightly
[image]

codingcatgirl, 2022-05-01:
Here's a project file that has the issue for me. [zhop_sample.zip](https://github.com/supermerill/SuperSlicer/files/8599849/zhop_sample.zip)

supermerill, 2022-05-02:
can reproduce only on ubuntu.

TravisWilder, 2022-05-02:
I can reproduce this with the predefined model "bunny" on a arm build from master-dev 2022-05-02 with the default settings from Voron 2.4 350mm
Also during slicing it looks fine at the very end of the slice the display changes to the error
[image]

supermerill, 2022-05-02:
I only fixed the gcode generation.

I'm on the verge of reverting all that to "please all use the '.' ", what do you think?

codingcatgirl, 2022-05-02:
For the Fan Speed: When superslicer is in german and i set the gcode preview to show me the fan speeds, it seems to ignore any  fan speed commands that set that set the fan speed to a non-integer value (like 127.5), and displays lines after that command as having the same fan speed as before the command. It confused the heck out of me and i when i saw the decimal point in the Gcode i realized it might have to do with locales.

For the settings. When i activate "simulate prusa no thick bridges" it sets two settings to something with a floating point, apparently. however, it's not displayed as such, it only shows the digits before the dcimal points. if i simply click save, often when i return to this page these values are shown in orange because they supposedly have changed even though i haven't touched them. only when i launched superslicer in english did i see that the values in those fields are decimal numbers, not just 55% as an integer.

supermerill, 2022-05-02:
Fan speed: seems fixed.
no thick bridges: can't reproduce. Can you send me a project where clicking on this setting produces the problem ?

[image]

codingcatgirl, 2022-05-02:
I don't think it's project related but i'll see if i can find a way to "export" the issue. But in your screenshot it shows a decimal number in the bridge line density numbers, which is something i haven't seen before so it might be fixed. I might just install the nightly tomorrow and see if it keeps happening.

closed 2024-02-11 by supermerill