supermerill/SuperSlicer #2774: Heavy memory usage

alexvonsiebenthal, 2022-05-01:
### What happened?

Superslicer eats lots of RAM on my M1 (Pro) MacBook Pro. 2 instances are running, one of which is empty. The empty one consumes over 2GB of memory, the one in used with some models loaded over 4GB.

Is this known behaviour? What eats all that memory? Can this be optimised?
[image]

### Project file & How to reproduce

just open and use SuperSlicer 🤷 

### Version

2.4.58.3

### Operating system

macOS 12.3

### Printer model

voron 2.4 / stealthburner

supermerill, 2022-05-01:
Is it a problem?

fusion also use many memory.

alexvonsiebenthal, 2022-05-01:
Yes, but fusion supposedly has to do lots more than superslicer :-) it is a problem in terms of overall resource management. I even had SuperSlicer instances crash (I think terminated by the OS for going over board with memory usage)

supermerill, 2022-05-02:
superslicer has to store each facet generated by fusion360, multiplied by the number of layers, multiplied by the number of perimeters, and also the paths for all infill moves.

I advise to use a `resolution` setting to avoid using models with too many details.

alexvonsiebenthal, 2022-05-02:
> superslicer has to store each facet generated by fusion360, multiplied by the number of layers, multiplied by the number of perimeters, and also the paths for all infill moves.
> 
> I advise to use a `resolution` setting to avoid using models with too many details.

Fair enough. However, it doesn't explain why an empty instance consumes over 2GB of RAM.

supermerill, 2022-05-02:
> Fair enough. However, it doesn't explain why an empty instance consumes over 2GB of RAM.

On my windows & linux, it's some mio.
I blame macos.
Maybe you're using a intel version on arm?

alexvonsiebenthal, 2022-05-02:
It is the intel version indeed (nightly). Might have fetched the wrong one then and will check with the ARM version.

supermerill, 2022-05-02:
the arm currently isn't buildable by servers. You have to compile it manually.
Here is the latest I built: https://mega.nz/file/NxIigRwR#hKgPdD1LykfLO3cwrUt_Qzs4BvcDVgh94_KDSNcV788

blalor, 2022-05-12:
As a point of comparison, PrusaSlicer 2.4.2+arm64 (running on my M1 MacBook Air) is using < 1G with a model loaded and sliced, and it's been open and idle for a day. 
[image]

supermerill, 2022-08-10:
[image]
For the record, my new dev arm version on arm Mac. Should already be an improvement.