supermerill/SuperSlicer #2775: Error

gin8330, 2022-05-01:
### Version
Version 2.3.57.12 

### Operating system type + version
Edition	Windows 11 Pro
Version	21H2
Installed on	‎31/‎10/‎2021
OS build	22000.652
### Behavior
Wan`t slice model. Slicing end up with error. "Cura" slicer do with no sweat. 
have attached all files you can tray it self.

_Is this a new feature request?_
Personalty for me it is difficult to follow Super Slicer versions... because it it not graphical visual presentation of versions. What is wrong just to make a simple numeric list with no "novel" like explanations.Only few from thousand people reads that. List should look like this:
1.Version 2.3.57.12 -date
2.Version 2.x.x.x-date
3.2.Version 2.x.x.x-date  and so on...  
As example: Prior Version 2.3.57.12 I have had Version 2.3.58.? And at this moment it is not possible
to find on your github...😒

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[1.zip](https://github.com/supermerill/SuperSlicer/files/8599718/1.zip)
[Alpair 5cone.zip](https://github.com/supermerill/SuperSlicer/files/8599727/Alpair.5cone.zip)

supermerill, 2022-05-01:
> As example: Prior Version 2.3.57.12 I have had Version 2.3.58.? And at this moment it is not possible
to find on your github...😒

https://github.com/supermerill/SuperSlicer/releases

you can also take the nightly that ahve some more fixes for the 2.4.58 branch:
https://github.com/supermerill/SuperSlicer/actions

gin8330, 2022-05-02:
Thank you,
would you be kind and explain how to use those fixes for the 2.4.58 branch:
https://github.com/supermerill/SuperSlicer/actions

Thank you in advance

supermerill, 2022-05-02:
When slicing your project, I have a message box 'error, there is an object with no extrusions on the first layer'

That because there is nothing to print on the first layer, a clear sign of problem.
If you're 100% confident it's not an error, there is a switch in the config to disable that error, but don't do that.
You should use the `place on face tool` and select the bottom of your object, so it's well sit on the build plate and not tilted.
If you don't want to rotate it, then I advise you to cut it with the cut tool so it will have a flat bottom.

gin8330, 2022-05-02:
Thank you for your explanation.Might be possible to make the program "snap"
objects to the printing plat in a same way.
Cura does that.
The other problem end of printing:
How to program end G code as the printed item would be "presented" on the
printing plate. Next puch "finish print" button.
Next, the printing head will be placed itself to the left side of painting
plate. But not to drive up on Z axes and stay there,that is pretty
irritating.

Thank you