supermerill/SuperSlicer #2777: Version Controlled Profiles/Filaments/Printers configs

piotrb, 2022-05-02:
**Is your feature request related to a problem? Please describe.**
I would like the Profile/FIlament/Printer config to be (perhaps optionally) be managed in a plain text folder that I can manage with version control (eg Git) .. so that I can easily see the changes to them and be able to easily go back to an earlier version if needed.

**Describe the solution you'd like**
I think all that would be needed is an option to choose a folder where configs are stored, and to allow them to be stored in some plain text format .. not sure if this is already possible ;) .. then I can just add git to the folder and we're golden.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2022-05-02:
if you create a directory named 'clibration' next to the exe, it's used instead of the default configuration folder you can see in `help -> show configuration directory`
you can also start the app with the parameter `--datadir="/data/myconf"`

closed 2022-05-02 by piotrb

piotrb, 2022-05-03:
lol this may have changed my life ;) being aware of every little bit of diff to those files is actually making it so much easier to not accidentally break my settings :)