supermerill/SuperSlicer #2780: Not all printer settings differences are marked

pa0los, 2022-05-04:
### What happened?

When a project has different printer settings then the current ones, some profile changes are not marked with orange color in the left tab. In this case the maximum acceleration was changed, but I think the same problem can be reproduced with other settings.
[image]
[image]

### Project file & How to reproduce

Can be reproduced with any project

### Version

nightly build

### Operating system

windows 10

supermerill, 2022-05-04:
can't reproduce, what are the steps to get that?

pa0los, 2022-05-04:
1. Create and save a project with some printer settings
2. In another project do modifications (like max acceleration) to the same printer profile and save the printer settings
3. Open again the previous project and compare settings against the profile changes

neophyl, 2022-05-04:
Correct me if I'm wrong but isn't that how its supposed to work ?  

So if I open a new project and select Profile A.  Then I change a setting , say retraction from -1 to -2mm.  The software stores in the project when I save, that retraction in this project is now -2.

When I reopen that project it still knows that retraction is different from the saved profile as the project stored -2 and can see that the profile its based on is -1 so that value is marked as different.

However if I go and open a different project and change retraction to -2 and then save the Profile over Profile A then profile A is now configured as retraction -2mm. 

If I then open my original project it loads the settings in, retraction is -2mm (as thats what is stored in the project) but it doesn't mark it as being different as what I have in my 3mf matches the profile now (as the profile was updated to also use -2mm)

In your example did the items not showing as different match your update to the underlying profile ?

supermerill, 2022-05-04:
as neophyl said.

some picture to illustrate:
unsaved change:
[image]
using a value different from the stored preset:
[image]

pa0los, 2022-05-04:
OK, then even the colors don't match.
See the screenshots in first post. The saved project has different Max acceleration from system setting. Both the category "Machine limits" and the setting name "Maximum acceleration..." are green.

supermerill, 2022-05-05:
> Can we change this behavior?

did you read my last answer? I'll copy the relevant part:

> You ask for a new colour for "current value is equal to the system preset, but different than the saved preset." and keep green only for "current value is equal to system & saved preset" ?

The orange takes precedence, but there is no orange as it's equal to the system preset.

> Or should I rather unlink the profile from "system preset"?

you can.

pa0los, 2022-05-05:
Yes, I read your answer.
I don't think another colour is necessary.
Just tried to explain that because of the order of "checks" colour could be misleading.
**iMO setting to green should be last on the list** (If other conditions are not met).
In the meantime I have unlinked the profole and it works as expected.

ristein, 2024-08-14:
In my Filament Settings Tab the categories are not marked orange when settings are changed. In Print ans Printer Settings Tabs this is working:
[image]
[image]
[image]