supermerill/SuperSlicer #2783: Inconsistent perimeter edges in thin walled model

fhteagle, 2022-05-05:
### What happened?

New-ish convert to SuperSlicer. Generally things are working well, but I have run into a bit of a head scratcher in SS behavior. Might be a bug, might be me running into an "undocumented feature", or my settings need adjustment. Search engines here and on the wider interwebs did not turn up a similar problem write up, nor solution.

When slicing this model: https://www.thingiverse.com/thing:4778778/files

I get very inconsistent placement of the start/stop edges in the thin upper part. See screenshot and attached gcode (in project .zip). 

Final results when I go ahead and print are incredibly under-extruded, layers not connected, etc.

Is this a bug? Is there a setting I need to tweak? 

Thanks for the assistance
- DH

### Project file & How to reproduce

[MasonBasil.zip](https://github.com/supermerill/SuperSlicer/files/8632445/MasonBasil.zip)

Steps to reproduce:
1. Download .stl from Thingiverse
2. Import .stl into SS
3. Slice with settings in project file
4. Note sliced layers preview output gives very jagged, inconsistent placement of edges all throughout the central "cone" part.

### Version

2.3.57.12 from Arch repos / AUR

### Operating system

EndeavourOS (arch linux based)

### Printer model

Creality CR-10 S5

supermerill, 2022-05-05:
it's too thin to print with a perimeter loop. enable "overlapping external perimeter"

fhteagle, 2022-05-05:
Thanks!

I had seen that option in the expert settings, but hadn't comprehended it's
function from the description. Not sure of how to rewrite the tip for that
option but I'll think about it and see if I find up with a better idea.

The slicing preview certainly looks better with that enabled.

Re-slicing and printing again now, I'll report if it fixes it completely.

fhteagle, 2022-05-06:
Printed the re-sliced file with this setting enabled, and the results were dramatically improved. Closing the issue, but I will keep thinking about how to improve the option help text. Thanks.

closed 2022-05-06 by fhteagle

supermerill, 2022-05-06:
Just a note: in the next version, this settings is changed and now is a percentage of the allowed overlap instead a switch between 0% and 100%.
So you may take that into consideration for the tooltip.

fhteagle, 2022-05-06:
Interesting evolution. I'm almost wondering if the options should be split three ways in a "thin wall handling" group:
1. Option 1, reduce extrusion width to match model as close as possible (with associated warning about possible under-extrusion and poor layer adhesion, advice to switch to a lower nozzle diameter, etc).
2. Option 2, thicken the "working copy" of the model the slicer uses to determine tool path as needed to give at least 1 full extrusion width (with associated warning about dimensional inaccuracy)
3. Option 3, allow or enforce at least the minimum perimeters setting as partially  thick passes (with associated warning about over or re-melting existing plastic).

No idea if these would be helpful and not too complex, but I figured I would share the idea that makes sense as an end user, and let you all figure out if they are worth implementing or not.

Thanks