supermerill/SuperSlicer #2787: Allow multiple layers before milling

Scissors95, 2022-05-06:
Currently the milling operation occurs at the end of each layer. This increases printing time with more tool changes and does not allow the tool to get a large enough chip load to clear properly. It works but the result is a surface finish that is suboptimal.

What would be handy is a setting to mill every X layers or mm This allows you to get some material built up before a milling pass is called. 

Behavior is similar to current except less frequent. A plunge may have to be added for milling internal features. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2022-05-06:
It's easy if there is no slope. What to do for thme? just ignoring them and mill at the position of the last layer?

Scissors95, 2022-05-06:
It would either have to ignore the sloped faces or mill every layer in that
case. At least in my use case I'm going to use the mill to ensure
dimensional accuracy on vertical faces.

It actually doesn't make much sense to use a square end milling cutter to
do slopes IMO and using a ball end to surface the slopes would be
incredibly labor intensive to add to the slicer and outside the intended
scope of this feature request.

fedonkin, 2022-09-27:
Hello. It would be interesting to have this addition :)

fedonkin, 2022-09-27:
When to expect a super supplement?)