supermerill/SuperSlicer #2788: Add used filament weight to the ruler

biddster, 2022-05-06:
In order to help me use up old reels, I weigh them. What I'd like to see from SS is the ability to use the ruler to pause the print when an amount of filament has been used. 

I can see that SS already shows what length of filament is used when you put pauses in using the ruler. But I'd like to use the ruler to put a pause in when I've used a certain amount of filament. 

BTW - Thanks for a really, really good slicer.

supermerill, 2022-05-06:
> I can see that SS already shows what length of filament is used when you put pauses in using the ruler

can you show a picture of the ruler showing the filament length? I'm not sure to understand.

KD7LRJ, 2022-05-11:
Maybe something like this?
[image]

biddster, 2022-05-11:
Exactly what I had in mind @KD7LRJ . Perfect representation of what I had in mind. Thank you.