supermerill/SuperSlicer #2793: Can't see latest versions

TheNewParad1gm, 2022-05-10:
### What happened?

The latest version I can see in releases is 2.3.56.9 (No more unwanted infill, on slopes). If I click on a link to 2.4.58.2 from an issue it takes me there, but going to the releases page by clicking the releases link from within Superslicer or on here only shows 2.3.56.9 and earlier versions.

Searching for version 2.4.58.2 also will not bring it up.

I was able to view later versions about a month ago when I last looked.

I have tried clearing my browser cache.

### Project file & How to reproduce

This is not related to a print/project file

### Version

2.3.57.12

### Operating system

Win 10

kpdcompany, 2022-05-10:
same, setting up a new computer from a crash, we need the latest ASAP, thats the only files we have for settings!

marcin76, 2022-05-10:
go to releases and klick on "TAGS".
or direct link to latest ver.
https://github.com/supermerill/SuperSlicer/releases/tag/2.4.58.2

supermerill, 2022-05-18:
seems like a github fluke.