supermerill/SuperSlicer #2795: Extrusion Multiplier is not working

Finest, 2022-05-11:
### What happened?

The Problem is when you add a few text calibartion cubes and set for each cube a differen extrusion multiplier, all cubes looks the same.

we tested this from 90 % up to 100 % all looks the same.

is this a know bug in the newest beta version ?

### Project file & How to reproduce

[Calibartion_cube.zip](https://github.com/supermerill/SuperSlicer/files/8673802/Calibartion_cube.zip)

here is my project how i configured the cubes with filament -> Extrusion Multiplier
and i set the filamen extrusion multiplier to 1 to see what the best flow is for my filament.

### Version

2.4.58.2

### Operating system

Windows 10

### Printer model

Voron 2.4r2

cruiten, 2022-08-19:
I noticed this same behavior but I am using version 2.4.58.4 on MacOS Monterey 12.5.

I thought that I was going crazy, but all the test cubes definitely appeared to be identical, despite the fact that each object clearly was assigned an individual extrusion multiplier value.

The-EG, 2022-12-09:
Pretty sure I'm also seeing this in 2.4.58.5. Trying to use the Filament Flow Calibration, all tests have the same results.

UnPure, 2022-12-18:
I'm currently on 2.4.58.3 and have the same problem, setting a per-object EM setting doesn't seem to work

AndrewEllis93, 2022-12-18:
Please upload your gcode files
We just discussed this in the Voron discord and it was indeed changing, just hard to see the differences for other reasons.

You can use the gcode previewer (move bottom bar left/right) and fine the same line in each cube. The E value will likely be different - which means it's applying.

charminULTRA, 2024-03-08:
Currently having the same issue on 2.5.59. How is this not fixed?