supermerill/SuperSlicer #2798: first layer 0.15 causes unexpected results

legend069, 2022-05-12:
### Version
_Version of SuperSlicer used goes here (help->about)_
Version 2.3.57.12 - the "copy version button " didn't copy the ".12" part at the end

### Operating system type + version
_What OS are you using, and state any version #s_
win10 pro _20H2

### Behavior
* _Describe the problem_ setting first layer height to 0.15 messes with more than just the first layer
* _Steps needed to reproduce the problem_ swapping back and forth between first layer 0.2 and 0.15
* _Expected Results_ only the first layer to be adjusted
* _Actual Results_ top layers and some perimeters get affected
  * _Screenshots if useful_
[image]
[image]

_Is this a new feature request?_ no
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here

[bug_report_Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/8677333/bug_report_Calibration.cube.zip)

_

neophyl, 2022-05-12:
Please read https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)
Specifically -
Strange pattern on a flat surface
A perfectly flat surface may not stay perfectly flat after being rotated in a 3D modelling software. If the slicer happens to slice across such a no more perfectly flat surface, weird one layer thick islands may show up. See [#1448](https://github.com/prusa3d/PrusaSlicer/issues/1448) and [#2434](https://github.com/prusa3d/Slic3r/issues/2434) for further details.

This is caused by vertices on the model having slightly different z heights and laying on the boundary between 2 layers.   
There's a few things you can do.  Check your model is flat to start with, very very slight rotations can cause it.  Second if its your model you can make doubly sure that the vertices are on the same plane to as many decimal places as you can.

legend069, 2022-05-12:
> 

Hi Neophyl, I'm not trying to be rude, but did you look at the project file and adjust things yourself?

as the bug happens from adjusting the first layer height, not by rotating it.

legend069, 2022-05-12:
wow you're right! thank you
I've been using fusion for almost 3 years and this is the first time I've encountered this problem,

looking at the rotation in fusion it has a rotation of "48.0000000000095781" before i use the align tool to align the face to origin.

using the tool "place on face" i would assume it first calculates the face angle to how ever many decimal place then it would rotate the model.

if the calculation rounds up/down or doesn't have enough decimal places then this would cause the problem everyone has raised.
i'm going to have a look into the code now to try to find it.

neophyl, 2022-05-12:
Its one of those continuous issues which is why it made it into the PS wiki on their github, but as SuSi is a fork then these sort of things are almost always common to both.

Sorry if I seemed to be cavalier with my original post, its just I've seen this lots, both on github but also over on the Prusa forum too so its one of those things you immediately spot.  

From following the various responses to previous issues on PS it looks like the place to face tool sometimes has issues with those very very slight rotations too.  Sometimes placing it to a different face and then back can 'fix' it and make it align.   Other times it can make it worse.  
Usually though when its baked into the object then the only thing left is to either tweak the layer heights in those areas using a height range modifier or to go in and fix the model.  I always prefer fixing the model though.

legend069, 2022-05-12:
"48.0000000000099973"  16 decimal places,
SS stores this as a double - 16 decimal places but it gets rounded to "48.00000000001" because of programming >:(
[stack overflow](https://stackoverflow.com/questions/9999221/double-precision-decimal-places)

my theory appears in SS when rotating by the distances above 
[image]

technically this is a bug, as it's just not possible to hold a rotation distance this precise then adjust it.(i wonder how cura does it, might be a python thing? ) cura doesn't make it as easy to find what things are called/defined as :(

i appreciate the help !
i'll try look into it over the next few days, if i find any decent data i'll post it here.

legend069, 2022-05-26:
closing, was not able to find any data in cura :(

closed 2022-05-26 by legend069