supermerill/SuperSlicer #2807: Infill/perimeters encroachment - inconsistent outcomes

TommyC81, 2022-05-15:
### What happened?

Using different settings for Infill/perimeters encroachment yields wildly different top layer results.

Example images:

0% encroachment:

[image]

10% encroachment:

[image]

15% encroachment:

[image]

20% encroachment:

[image]

25% encroachment:
[image]

### Project file & How to reproduce

Open attached project file, change "Infill/Perimeters encroachment" setting - observe top layer differences.
[Testpiece20.zip](https://github.com/supermerill/SuperSlicer/files/8694437/Testpiece20.zip)

### Version

dev_nightly (2e11c485266cf22987ae0f7be395df6a1295148c)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

Sirnut, 2022-05-17:
I have been noticing issues like this as well, but on all solid surfaces like this:
[image]
None of this started until I began tweaking the infill encroachment values, because I realized my horrible overextrusion on solid infill only was somehow related to it
I've gone ahead and attached the project file I'm working on at the moment. Please let me know if I should open a new issue, I just somehow feel like these are related
[Solid infill bug.zip](https://github.com/supermerill/SuperSlicer/files/8711303/Solid.infill.bug.zip)

supermerill, 2022-05-18:
@Sirnut
i don't see anything in common.
if you want the top fill to be larger, increase the default anchor.

Sirnut, 2022-05-18:
@supermerill sorry to bring this up again, but is the solid fill in this area not inappropriate?
Layer 22:
[image]
Layer 23:
[image]
Layer 24:
[image]

In layer 23 you can see where there is gap fill and solid infill where the sparse infill should be placed

supermerill, 2022-05-18:
that's not that issue. See #253 #1090

Sirnut, 2022-05-19:
I'll look into that. Thanks Merill

dehmlowm, 2022-12-06:
Thank you Super Merill. I noticed that this issue is about 50% better since upgrading to 2.5 ( its been a while since I upgraded). 

Wish there was a complete fix appreciate your contributions~

closed 2024-02-11 by supermerill