supermerill/SuperSlicer #2811: "Export Plate" (STL) does not export negative volumes part of the object.

TommyC81, 2022-05-16:
### What happened?

Exporting the plate (STL), expected the final object/plate contents (including negative volumes) to be exported. However, the negative volumes are not exported.

Piece:
[image]

Exported STL:
[image]

### Project file & How to reproduce

Open attached project file, go to "Export" -> "Export Plate".

Notice that the negative volumes are not exported to the STL (only the plain box).

[Testpiece20.zip](https://github.com/supermerill/SuperSlicer/files/8698034/Testpiece20.zip)
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/8698037/Shape-Box.zip)

### Version

nightly_dev (2e11c485266cf22987ae0f7be395df6a1295148c)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

neophyl, 2022-05-16:
That's expected operation.  Modifiers effect the slicing parameters they do not edit or change the underlying geometry of your models/objects.  If that's what you want then use a CAD program to do a boolean cut.

TommyC81, 2022-05-16:
> That's expected operation. Modifiers effect the slicing parameters they do not edit or change the underlying geometry of your models/objects. If that's what you want then use a CAD program to do a boolean cut.

Ah, I see. That makes sense, I'll close this one.

closed 2022-05-16 by TommyC81

jvandenbroek, 2023-12-18:
Since this is implemented in PrusaSlicer (https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6), is there any chance to get this ported over to SuperSlicer? For small changes it's way easier to do it quickly in the slicer than using a full blown CAD program.