supermerill/SuperSlicer #2814: No wipe occurs at end of layer despite crossing perimeter as next move

TommyC81, 2022-05-16:
### What happened?

No wipe occurs at end of layer despite crossing perimeter as next move.

This causes stringing - significantly so depending on the model and circumstances - when the start of the next layer requires perimeter crossing travel.

Ideally, a wipe could be forced at the end of the layer - before (optional) retraction and z change.

Example below, note that one perimeter has just stopped printing. Next move now will be a z change, then followed by retraction, then move to the new object/next layer start. This should ideally have been; wipe/retract, then z change (if not also z change whilst wiping), then move to the new object/next layer start.
[image]

STL test file (based on using 0.6mm nozzle): [TestPrint2 - Wipe and retract - STL.zip](https://github.com/supermerill/SuperSlicer/files/8702430/TestPrint2.-.Wipe.and.retract.-.STL.zip)

Project file: [WipeAndRetract_Test1.zip](https://github.com/supermerill/SuperSlicer/files/8702434/WipeAndRetract_Test1.zip)

Outcome in test STL (note stringing exclusively from where the previous layer ends to where the next layer starts - there is no stringing from finishing the first object then a proper wipe/retract on the same layer before continuing on the next object:
[image]

[image]

### Project file & How to reproduce

Open the attached project file:
[WipeAndRetract_Test1.zip](https://github.com/supermerill/SuperSlicer/files/8702457/WipeAndRetract_Test1.zip)

Slice and note that there is no wiping at the end of the layers, prior to the z change (after retraction thereafter) and travel to start the next layer on the other object. 

### Version

nightly_dev (2e11c485266cf22987ae0f7be395df6a1295148c)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

TommyC81, 2022-05-18:
In addition to this missing wipe, there's also no wipe occurring after a gap fill. This is somewhat negated if you don't do a z hop after retraction.

jeedee23, 2022-05-23:
There is an error in your file since 'thin wall perimeter' is set to 0%

TommyC81, 2022-05-23:
The 'thin_perimeters' setting set to 0% should merely disable the allowance for perimeters to overlap itself. The same apply for the 'thin_perimeters_all' setting. Changing these values doesn't change the (missing) wipe at all.

[image]

Not sure if you're using the same version, this is the gcode I have: 
[WipeAndRetract_Test1_gcode.zip](https://github.com/supermerill/SuperSlicer/files/8756805/WipeAndRetract_Test1_gcode.zip)

To note that a wipe (and retract) occurs nicely on the last tower on the final layer - the same wipe does however not occur on any layers prior leading to stringing.