supermerill/SuperSlicer #2815: Brew Install

Dragoid88, 2022-05-16:
### What happened?

When I install superseder through brew I gat a popup window saying the file is damaged
[image]

### Project file & How to reproduce

Run the brew install command for superslicer and try and open it

### Version

2.3.57.12

### Operating system

macOS 15

supermerill, 2022-05-18:
mac arm?

Dragoid88, 2022-05-18:
No, M1 13 Pro

Sincerely,
Andrew Johns

supermerill, 2022-05-18:
m1 is arm, right?

check that one https://github.com/supermerill/SuperSlicer/issues/2055#issuecomment-1126989720