supermerill/SuperSlicer #2817: Premature Calculation

Durahl, 2022-05-17:
### What happened?

Despite `Background Processing` being disabled in the `Configuration` > `Preferences...` > `General` Section...

Applying any changes to the values in the `Settings for Height Range` after applying a `Height Range Modifier` to a part will _immediately_ reset any already generated Slices despite not having made any changes to it yet - All I've told it is that I'm about to apply a change in a particular region - I haven't told **SuperSlicer** what that change is which is kinda _**SUPER™**_ annoying _especially_ on Plates that take a while to slice and where I usually end up wanting to add multiple Height Range Modifiers to various Parts in one go while still having access to the previously generated Slicing Information to pass Judgement where to place the Height Range Modifiers.

On a related sidenote - May I suggest that a newly generated Height Range Modifier has its `Start at Height` value automatically set to the currently viewed Height? That would be Super Nice 🤩

### Project file & How to reproduce

1. Create a Cube ( or a Cylinder - I won't judge... 🤞 )
2. Slice the Cube
3. Apply Height Range Modifier to Cube
4. Change any one of the Starting or Ending Height Range Values - Confirm with **_Enter_**
5. Cube / Build Plate will lose all previously generated Slice Information - ***POOF***
6. Find nearby Corner - Start Crying... 😭

### Version

2.4.58.2

### Operating system

Windows 11

### Printer model

Voron 2.4 with StealthBurner & ERCF

supermerill, 2022-05-18:
you can't do that on the slicing preview instead of the gcode preview?

Durahl, 2022-05-18:
I'm referring to how a `Preview [Ctrl+3]` is being reset to `3D editor view [Ctrl+1]` with all Slicing information `[Ctrl+R]` lost that was previously generated to navigate the `Preview [Ctrl+3]`.

Like when I'd would want to add a different Top Infill for _every other_ Step on a Staircase - The obvious choice is to set a default Top Infill for the _odd_ Steps in the `Print Settings` and then add `Height Range Modifiers` for every _even_ Step. The easiest way to do that is with the "Ruler" on the Sidebar available but that too vanishes in the Slicing Data reset.

I recall the behaviour until "recently" to have been that the Slicer would not erase all previously generated Slicing Information while I was adding multiple Height Range Modifiers, changed Values, defined Modifiers, etc... _Until_ I actually wanted a new Preview generated by hitting the `Slice now [Ctrl+R]` button.

I also _just_ noticed there being no difference between `Window` > `Layer Preview Tab [Ctrl+2]` and `Window` > `GCode Preview Tab [Ctrl+3]`?

supermerill, 2022-05-18:
> I also just noticed there being no difference between Window > Layer Preview Tab [Ctrl+2] and Window > GCode Preview Tab [Ctrl+3]?

The slice preview is what's generated by the slicing (some colors aren't exact, mostly for overhangs)
The gcode preview is a view recreated from the exported gcode file.

For your problem, when a setting is modified, the slicing is reset just before its first use.
If you are changing the regions / the slicing height, then the very first step has to be reset.

Height range modifiers is designed to be set / added in the 3D preview, as it's applied on the input geometry, not on the gcode output.
The most problematic feature is the custom gcode/color change that need one of the two preview to be set, but invalidate the gcode one.

I understand the frustration. What could be done is a kind of checkbox that can "freeze" the preview, while deactivating the slice/export button, so you won't export/slice somethign that is out of sync.

Durahl, 2022-05-19:
Are you sure the Height Range Modifier is meant to be used in the 3D Preview? 🤨

From my P.o.V. it makes absolutely no sense to do it there - or without access to the Sliced Preview ( which gets constantly reset ) because for one - in the 3D Preview you're basically playing a guessing game what values you need to set a HRM to even just get it into the Ballpark where you want it as you can't just slide the Planes up/down ( unless you have memorized the dimensions of **your** parts ) whereas in the GCode Preview you're at least able to go to the _exact_ Slice you want to get the Value you need ( hence my addendum of being able to have a HRM automatically generated at the currently viewed GCode height ) and for the other you're going to correct it anyway because your off by one slice - I cannot stress how odd it is to add a PAUSE command ON the Layer that _closes_ a Hole instead of while it is still open - Very unintuitive choice from Prusas side 😑

I'm not quite sure how I'm supposed to understand the _Freeze_ Option - The way I interpret it nothing changes that is causing my frustration, but it instead just prevents me from Exporting? 🤨

supermerill, 2022-05-19:
> I'm not quite sure how I'm supposed to understand the Freeze Option - The way I interpret it nothing changes that is causing my frustration, but it instead just prevents me from Exporting? 🤨 

It prevent any reset and any exporting. When deactivated, it will reset everything and allow for slicing/exporting.