supermerill/SuperSlicer #2821: Make custom bed shape respect the stl used, not only outer limits

Takryn, 2022-05-18:
I have imported a custom bed model (to both the "shape" and "model" sections of the bed shape definition) to account for a large damaged section in the center of the print area. While it is visually evident when moving prints around on the bed manually, when using "Arrange" or slicing it uses the entire print surface. 

The issue seems to be the bed "shape" only uses the perimeter/outer limits and assumes the center is continuous, even if the model used to describe it is not. When loaded under the "custom" dropdown for bed shape, the settings default back to rectangular 255 by 255 (which are the correct outer limits of the custom stl) When loaded under the bed "model" section, it accurately represents the non-continuous model, but has no impact on the printable area as used by the "Arrange" feature nor the slicer itself.

requested solution: allow creation of truly custom bed shapes, respecting holes in the model used to define the bed shape. modify/update the arrange feature and slicer programming to avoid the holes/properly throw "outside the build area" warnings with respect to the bed shape stl.

in the image below, the same stl was used in both "custom" shape and load "model"
[image]

in the image below, the arrange function was used, but it placed items over the hole in the print bed model
[image]

neophyl, 2022-05-18:
This request has been made on the PS github too https://github.com/prusa3d/PrusaSlicer/issues/3866

supermerill, 2022-05-18:
the model is just for the look. It's the shape (the top setting) that has a real inpact. But still only its bounding box is used currently.
Now that i think of it, i guess it's possible to add "fake static object" into the arrange function to force it to leave these areas alone.
Another problem is the way the shape is done: it's only a polygon, but you may want to add holes into it.

Takryn, 2022-05-18:
Cheers, thanks for looking into it!