supermerill/SuperSlicer #2825: Feature request/idea

inspectionsbybob, 2022-05-20:
### 
post 2.3.57

### Operating system type + version
All

### Behavior
Desired is a VERTICAL SLICE VIEWER.
This is to aid in viewing INFILL and related internal structures on complicated models.  
Often when building complex models with lots of recesses it is important to be able to see what the support structure looks like UNDER those recesses, but for successful printing, but also structural strength.
It is possible to scroll up/down using the layer preview, but that is difficult/slow on complex models, and sometimes with complex infills it is not clear what you are looking at.

This feature would TURN the slice preview 90deg so you could see the internal structure under a recess (or any element) at a glance.

This feature would NOT allow for scrolling horizontally through the model as the processing required would be to great, and slow. 

while in preview mode, you would select he "vertical slice" feature and then drag a "cut line" to the desired location. When happy that the line goes through the desired location the user would select "cut" and the system would generate a cutaway view at that location.  If a different view is desired, they would have to got back to the full preview screen and start again.

The cutline would be fixed at a straight front to back line, the preview model would be moved/rotated using the existing tools.

_Is this a new feature request?_
YES

jeedee23, 2022-05-21:
I think what you are asking is almost impossible, since this is not a 3D model, merely a drawing of vector shaped objects determined by the slicing parameters, If one could generate a 3D model of all the layers, then it would be possible, but the computing behind that is immense. What you are looking at is a representation of the gcode, so slice it vertically is impossible, at least I think.

Dx-Dx-Dx, 2022-05-27:
This is very useful feature. Especially for supports. One of the last things Simplify3D is still useful for.
Here is the screenshot of superslicer's gcode previewed in Simplify3D with cross section view enabled.
[image]

inspectionsbybob, 2022-05-27:
Yup, thats what I am looking for...

 found a few other current slicers, some of them didnt like the gcode from
SS, dont know if it was the gcode or the previewer,  but I reported it to
the programmers along with the file that made it upset.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

jeedee23, 2022-05-28:
> This is very useful feature. Especially for supports. One of the last things Simplify3D is still useful for. Here is the screenshot of superslicer's gcode previewed in Simplify3D with cross section view enabled. [image]

OO, didn't remember, I have a 10 y old lifetime license, so yes S3D is good for some aspects.

inspectionsbybob, 2022-09-25:
Touch to make current again.

inspectionsbybob, 2023-01-10:
touching again to bring it into 2023

inspectionsbybob, 2023-04-24:
touching again.

PLEASE pull this in!