supermerill/SuperSlicer #2829: SuperSlic3r logic problem between the speeds of the overhangs and bridge

Henyckma-Sensei, 2022-05-21:
### What happened?

It seems superslic3r sometimes differs in logic between overhangs that need or don't need bridge speeds

I really don't know how to explain it well... but some bridges take the speed of overhangs

It doesn't seem to be a critical problem, it can even be solved with a little support under the bridge

[image]

### Project file & How to reproduce

Steps to reproduce:

1: open the archive
2: select view type: Speed
3: see the problem

[PIEZA_LASER1_4.zip](https://github.com/supermerill/SuperSlicer/files/8748155/PIEZA_LASER1_4.zip)

### Version

2.54.8.2

### Operating system

Windows 10

### Printer model

Prusa i3 DIY

jeedee23, 2022-05-22:
There is not a version 2.54... we are now at 2.3.57.12. First update and then see if you have this problem. 
The blue on this screen is the speed of 20 mm/s, it is NOT the same blue if you change the view to features type, which is logical. You need to post the gcode, where one can see if this blue you indicate is overhang or bridge:
;TYPE:Overhang perimeter 

https://user-images.githubusercontent.com/38980341/169713579-5a5338d6-7664-477a-bb2f-bda8bdca0efd.mp4

;TYPE:Bridge infill
;TYPE:Internal bridge infill

Henyckma-Sensei, 2022-05-23:
> Im sorry, it was certainly my writing error, I already corrected the version although the program tells me version 2.4
[image]

yes, it is exactly the problem that I mentioned, it should be a bridge perimeter however, it takes it as an overhang causing the speed to drop and there is no way to maintain the speed, perhaps modifying the gcode manually or changing the overhang speed to equals te bridge speed 

[image]
[image]

The gcode:
[PIEZA_LASER_1_4.zip](https://github.com/supermerill/SuperSlicer/files/8755042/PIEZA_LASER_1_4.zip)

neophyl, 2022-05-23:
We certainly do have 2.4.58.2, its been available through the nightlies for awhile now.  Its also available if you display all tags as its a pre -release version.

jeedee23, 2022-05-23:
Hi, I have this same problem, the overhang speed (add_extra_perimeters on overhang) differs: it perfectly makes the overhang at 'outer perimeter' speed, but the other overhang_speeds are at internal perimeter speed (twice as fast in my case, so it drops)
[202205232143.zip](https://github.com/supermerill/SuperSlicer/files/8757636/202205232143.zip)
For your info, see the attached. Please give a link to this latest version?
In another post I mentioned using perimeter_width_even and perimeter_width_odd. It is a very interesting improvement for fucnctional obejectr which need more strength. I wrote a small script which slices first the object in 'even' mode, and then in 'odd' mode. Then I 'merge the two files, resulting in the 'merged.gcode' (so layer 0 comes from 'even', layer 1 from 'odd' etc)

But this overhang speed, I can not resolve it.

jeedee23, 2022-05-23:
Just downloaded this latest version. 
[image]
look at the red numbered dots:

1. External perimeters speed: that is the speed of the external overhang, depending on the acceleration 
2. Internal perimeter speed is the speed of the extra perimeters in your overhang
3. This version includes a percentage for speed on overhangs (Great! but...) , **still external perimeters and internal perimeters will differ, have tot test this**
4. You can play now with the acceleration on overhangs, could be very very useful, because, the longer and heavier your overhang gets, the faster the nozzle should move, nit also extrusion should then be linear with the acceleration.

Tip to the coders: just one base perimeter (external or internal) speed on overhangs please?

Henyckma-Sensei, 2022-05-25:
> Just downloaded this latest version. [image] look at the red numbered dots:
> 
> 1. External perimeters speed: that is the speed of the external overhang, depending on the acceleration
> 2. Internal perimeter speed is the speed of the extra perimeters in your overhang
> 3. This version includes a percentage for speed on overhangs (Great! but...) , **still external perimeters and internal perimeters will differ, have tot test this**
> 4. You can play now with the acceleration on overhangs, could be very very useful, because, the longer and heavier your overhang gets, the faster the nozzle should move, nit also extrusion should then be linear with the acceleration.
> 
> Tip to the coders: just one base perimeter (external or internal) speed on overhangs please? In the video: you see the first (external perimeters) perfectly take the overhang, and then the second time it passes, much faster and the extrusion false true
> 
>  2022-05-23_22-05-57.mp4

rather I think that you are taking the bridging speed (50mm/s), you can confirm it in the history speed section by modifying the value (bridge speed)

I have not yet confirmed if it is a problem with this version, but I think not since apparently what is missing are values ​​to be defined, for example isolating the perimeter of the bridging to separate them from the overhangs