supermerill/SuperSlicer #2834: Add a new language

AntoszHUN, 2022-05-24:
So I did the first complete 100% Hungarian translation for a program called SuperSlic3r. A little more low-cost.

  - This translation is only the first version.
  - There is still room for improvement.
  - Translation and spelling errors may occur.
  "I did it all by myself, so it's up to me."
  - I am constantly checking and improving the translation.

I have attached the zip folder in which the necessary files are included. I hope someone can unite. I can't upload a folder.

Thanks!

[hu.zip](https://github.com/supermerill/SuperSlicer/files/8765619/hu.zip)

AntoszHUN, 2022-06-11:
hmmm.

supermerill, 2022-07-12:
merged, will be available in next nightly/release.
note: if the english text has '%%', that means it need to be doubled to output a '%', or it will crash.
same for '%1%%%'

closed 2024-02-11 by supermerill