supermerill/SuperSlicer #2835: 2.3.57.12 Macos Arm is damaged and cannot be opened

kaisarantonn, 2022-05-24:
**Bug description**
Cannot be opened on macos 12.3.1 on Macbook Pro (M1 Pro)

**Expected behavior**
Working as smooth as prusaslicer

**To Reproduce**
Steps to reproduce the behavior:
1. Download [SuperSlicer_2.3.57.12_macos_arm_220401.dmg](https://github.com/supermerill/SuperSlicer/releases/download/2.3.57.12/SuperSlicer_2.3.57.12_macos_arm_220401.dmg)
2. Click on the downloaded File
3. Copy Superslicer into Application on Mac
4. See error "“SuperSlicer” is damaged and can’t be opened. You should eject the disk image."

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 12.3.1 Monterey

stephbellemare, 2022-05-24:
See this comment as it's already documented
https://github.com/supermerill/SuperSlicer/issues/2055#issuecomment-1126989720

mattheath, 2024-05-11:
Specifically this: https://github.com/supermerill/SuperSlicer/issues/2055#issuecomment-1195640131

After copying it to /Applications, you need to run the following:

```
xattr -cr /Applications/SuperSlicer.app
```

Jmack66, 2025-04-16:
Just a further note to anyone else who runs into a similar issue and doesn't daily macos: 

For me on my m2, the dmg would not show the applications folder when clicked (whereas all other dmgs show the application folder to drag the app into), dragging app inside the dmg onto my desktop and then into applications allowed me to successfully run the above command with sudo