supermerill/SuperSlicer #2836: Missing retraction on first layer

MarkoAnte, 2022-05-25:
### Version
2.4.58.2

### Operating system type + version
Windows 10

### Behavior
* Missing retraction on first layer
* Make a box with cut out and toggle "Only retract when crossing perimeters" disabled
* First layer should have the same status at top layer
* No retraction happens when not crossing perimeters, making the extruder leak on long moves on large objects, leaking blobs on first layer. Those get printed over resulting in imperfections on bottom of parts.
  * Missing Retraction
[image]
  * Retraction forced with "Only retract when crossing perimeters" disabled. 
   [image]

_Is this a new feature request?_
I think this is a bug. 

If its not a bug, it would be nice to have an extra option "Only retract when crossing perimeters" = Disable for first layer only. 

#### Project File (.3MF) where problem occurs
[RetractionFirstLayerProblem.zip](https://github.com/supermerill/SuperSlicer/files/8769816/RetractionFirstLayerProblem.zip)

MarkoAnte, 2022-05-25:
I checked in PrusaSlicer. 

[image]

This is with "Only retract when crossing perimeters" enabled. So I think its more of a bug then a feature to be.

tome9111991, 2022-05-26:
could be true.
maybe related with
https://github.com/supermerill/SuperSlicer/issues/2802

fungreenfox, 2022-08-15:
Is it solved? This issue is still present in version 2.4.58.4

tome9111991, 2022-08-15:
> Is it solved? This issue is still present in version 2.4.58.4

you have a checkbox now to enanle retract on first layer

closed 2024-02-11 by supermerill