supermerill/SuperSlicer #2837: Bridge flow ratio only has an effect when bridge flow baseline is set to nozzle diameter

Jamin82, 2022-05-25:
### What happened?

Trying to adjust bridge flow when i had similate prusa "no thick bridge" selected I noticed that no matter what setting I entered (tried all the way from 5% to 500%) the lines in the gcode visualiser would be the same width and the used plastic would stay the same.
this was also the case when "keep current flow" or "layer height" was selected as a baseline. without simulate prusa option selected.
When nozzle diameter is selected as baseline bridge flow responds correctly. Lines change width and printed part weight changes

keep current flow:
[image]
[image]
[image]
[image]

layer height:
[image]
[image]
[image]
[image]

Nozzle diameter:
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[bridge flow 100mms.zip](https://github.com/supermerill/SuperSlicer/files/8772419/bridge.flow.100mms.zip)
[image]

### Version

Version: 2.4.58.3 Build: SuperSlicer_2.4.58.3_2022-05-18

### Operating system

win 10

tome9111991, 2022-05-26:
look at volumetric flow view?
[image]

Jamin82, 2022-05-27:
> look at volumetric flow view? [image]

using that view as you suggested confirms the issue that i brought up
here is using 5% flow
[image]

here is 500% flow
[image]

jeedee23, 2022-05-28:
Yes I searched a lot for that, especially to have a good result.  Speed, temp, distance between the lines, etc all are in play to have a good result (result differs from view). If I gave you my settings, maybe they won't work with you. Difficult, but doable. I don't use the 'prusa' option. Just nozzle diameter , temp, speed and distance between lines are important. (this for Abs)

Jamin82, 2022-05-28:
I have inspected even the gcode and the flow does not change unless you have selected nozzle diameter as the flow reference.

This is not about me looking for good print settings. This about functionality that is not working properly within the slicer

jeedee23, 2022-05-28:
Can you send both project files? (save as ...3mf or ...amf) I tweak myself through the gcode with my own scripts afterwards, in order to achieve good results, but  it could be that I am inventing the warm water...

jeedee23, 2022-05-28:
you can use this, SS will revert (as in many settings, in our favor) 
[image]

Jamin82, 2022-05-28:
Have you read any or my issue that I filled out above.. I get the same extrusion rates at 5% as at 500%

Jamin82, 2022-05-28:
Also when you use nozzle diameter it is possible to go above 100% as long as you use nozzle diameter as the flow reference.

So to summarise my whole reason for opening this issue:
Flow ratio behaves as expected when you use nozzle diameter as bridge flow reference.
It does not work as expected when you use keep current flow or layer height as flow reference

jeedee23, 2022-05-28:
[image]

I think I understand what you mean now, but is that an issue?

jeedee23, 2022-05-28:
[image]

This is what you mean: so the values  below in the recatangle shoud be greyed out when choosing 'keep current flow'
[image]

jeedee23, 2022-05-28:
also in this case:
[image]

jeedee23, 2022-05-28:
To confirm my 'saucage' reasoning:
[image]

jeedee23, 2022-05-28:
but possible if you choose nozzle diameter, resulting in grinding extruder, here are your saucages:
[image]

jeedee23, 2022-05-28:
> Also when you use nozzle diameter it is possible to go above 100% as long as you use nozzle diameter as the flow reference.
>
> So to summarise my whole reason for opening this issue: Flow ratio [...]

Please accept my correction of your statement:

Flow ratio behaves as expected when you use nozzle diameter as bridge flow reference. It does not work, as expected, when you use keep current flow or layer height as flow reference, so the 'flow ratio' settings should be greyed out.

Do you agree on that?

Jamin82, 2022-05-28:
No I do not agree with that at all. If I wanted to not be able to tune individually all parameters of flow I would use prusaslicer.
At its core superslicer is about having fine control of all parameters of your slicing.
If I want my bridging to be calculated off of layer height but then I would like a little more or less flow the bridging multiplier should work in the same fashion as it does when I select nozzle diameter as the basis for calculation

jeedee23, 2022-05-28:
I think all of my above example show that you can indeed have fine control over bridging. If you want layerheight, then you can't adjust that because there is nothing below or above, so it's a cilinder with the diameter of the layerheight.  If you want the current flow as baseline, it will follow the flow you indicated in your settings. The only way to tweak and or refine is like I show here above.
The problem is that you think the settings can have influence where they can not, and that should be approached by greying them out.. You can tune anything you want, but you don't follow the logic.

Jamin82, 2022-05-29:
Following your logic we should not be able to adjust bridging flow when it's selected as nozzle diameter. Because then it would no longer be the nozzle diameter

jeedee23, 2022-05-29:
No you are wrong: you can choose 3 baselines :

1.  nozzle diameter (in your case 0.4 mm) you can increase
2. layer height  (in your case 0.2 mm) you can only decrease
3. keep current flow.  (in your case the perimeter flow of 0.48 mm * 0.2mm * 1mm = appr 0.1mm³/mm) you can only decrease
It is all very well (extremely well) explained in the pop-up help:
[image]

Jamin82, 2022-05-29:
OK.. But why when I reduce flow on layer height or keep current flow does it have no effect?

jeedee23, 2022-05-29:
you are right there:
[image]
[image]

jeedee23, 2022-05-29:
conclusion: there should be only one option present: nozzle diameter. the rest is obsolete

Jamin82, 2022-05-29:
So you now finally understand why I named this very issue **Bridge flow ratio only has an effect when bridge flow baseline is set to nozzle diameter**

All the other points you have tried to discuss have nothing to do with the entire reason i opened this issue. It has taken you all these messages to finally get on to the target that i **CLEARLY** outlined in my very first post. That flow does not have en effect when you set the calculation based on layer height or current flow.

> conclusion: there should be only one option present: nozzle diameter. the rest is obsolete
Obsolete? You do realize that these additional options for flow calculation are brand new in the 2.4 beta build? And the fact that they are so new easily explains why the bridge flow modifier was likely overlooked in this new implementation (Again this is why i opened this issue to help point out this small bug in the brand new implementation of flow calculation)

[image]
I tried to point this out multiple times to you. But you insist on highlighting text from the tooltip for some strange reason. which again has nothing to do with the title and intent of the issue 

and not only that you have contradicted yourself multiple times in this thread:

first you say that i can only reduce flow when set to options not nozzle diameter
[image]

then you say i cant change flow
[image]

Then you say i can only reduce flow again:
[image]

jeedee23, 2022-05-29:
You are right, but I used the tooltip as a base for my explanation.  But I remain with the fact that both other options are obsolete, if you can adjust the amount of bridging flow, speed and density, there is enough. A useful option would be: the longer the bridge the more the pressure advance system should work and the acceleration should be adapted. 
Anyway, thanks for your response and intelligent remarks. I think the programmers have enough to optimize this overtime.
Could you give your remarks to my opinion on this (another thing): https://github.com/supermerill/SuperSlicer/discussions/1800#discussioncomment-2842728
have a nice day!

supermerill, 2022-07-12:
Sorry, i didn't read the psot, as the first one was enough to find & fix the bug. If there is another problem, please tell me.

jeedee23, 2022-07-12:
It's ok, just check my question about even and odd layer perimeter width option please. If odd-even perimeter widths can change, structural behaviour is much better.

closed 2024-02-11 by supermerill