supermerill/SuperSlicer #2839: Internal bridge infill - extreme over extrusion and no apparent setting to adjust

TommyC81, 2022-05-26:
### What happened?

Supporting dense layer (anchored) itself works great - it's anchored proper and prints nicely.

[image]

[image]

However, the "internal bridge infill" that is printed on top is extremely over extruded (plastic sticking up 0.2-0.3mm above the current layer, a sea of plastic and significant nozzle hits).

[image]

There are no apparent settings for this "internal bridge infill" layer, and it doesn't use the "solid infill" settings. Hence, there is no readily available possibility to change this.

If I don't use a supporting dense layer (anchored), then the first "internal bridge infill" layer prints significant portions in the air, leading to unpredictable outcomes.

[image]

It would be great to have an option to simply go straight to the normal "solid infill" on top of a supporting dense layer, or at least have options available to configure the "internal bridge infill" immediately on top of the supporting dense layer.

I'm unable to 100% determine whether the over extrusion is caused by the supporting dense layer being extra thick, or because of the "internal bridge infill" on top being extra thick, or a combination thereof.

### Project file & How to reproduce

[Testpiece41.zip](https://github.com/supermerill/SuperSlicer/files/8776682/Testpiece41.zip)

Load attached project file.
Slice.
Note the dense/anchored infill being prepared nicely.
Note the "Internal bridge infill" on top of the supporting dense layer.
Note that internal bridge infill on top of the supporting dense layer has no apparent settings.
Change settings and notice no changes to the internal bridge infill.

### Version

nightly_dev (65f41c83000dc3e9c54656fe074f1963fd4e34ba)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

neophyl, 2022-05-26:
You mean like this -
[image]

The supporting dense layer was added just for the problem of that first layer sagging into all the gaps of the normal infill.  By turning it off you basically get the behaviour of Prusa Slicer.  So the fact that bridge infill does that isn't so much a 'bug' as allowing users to have the same behaviour as PS, if for some reason they really really wanted to.  Personally I can't think of a good reason but hey there is always someone out there who can :)

Attached the project I used for the screen cap.

[Testpiece41_neo.zip](https://github.com/supermerill/SuperSlicer/files/8777640/Testpiece41_neo.zip)

TommyC81, 2022-05-26:
> You mean like this -
> [image]
>
> The supporting dense layer was added just for the problem of that first layer sagging into all the gaps of the normal infill.  By turning it off you basically get the [...]

I am using it the dense infill layer for that purpose. That's not the problem. The problem is that it causes massive over extrusion and there's no obvious way to configure the internal bridge infill (along with the dense supporting layer underneath) to fix this. In short: the available choice is seemingly a) sagging infill (dense infill layer off), or b) significant over extrusion (dense infill layer on).

The screenshots and description in my original post should capture this. Let me know if it is unclear.

neophyl, 2022-05-26:
The project I posted does address this.  As you can see that bridge infill layer is thinner than on your original.  You have the default assumption configured that bridge infill is nozzle diameter which means its 0.4 high.  That would equate to around the 0.2 sticking up as you put it.  If you change the assumption via the setting then it doesnt do that any more.

TommyC81, 2022-05-26:
Oh, I see, that's probably the setting I've been looking for. Maybe the
different terminology makes it hard to find. I'm away from computer now,
but will try this and report back.

To note that I have another open ticket that is somewhat related, and this
hasn't been suggested before. This might solve them both. Thanks!

TommyC81, 2022-05-28:
To follow-up; confirmed that the setting suggested by @neophyl was indeed the issue. It was no apparent that it was the applicable setting. As such, the issue relates more to the visibility of the setting and finding it rather than a bug. I will close this ticket to not clutter the tracker.

[Testpiece42.zip](https://github.com/supermerill/SuperSlicer/files/8790733/Testpiece42.zip)

closed 2022-05-28 by TommyC81

YahyaAbderrafai, 2023-10-06:
Follow up: option to deactivate should be available.

The premise that a "bridge line cross section is circular" is wrong when applied to high infill. When applying the layer height baseline, you end up with very thin width. I have a case where the nominal width is 0.72 (0.6mm nozzle) and height is 0.2. So the bridge width is 3 times as small as the nominal width.