supermerill/SuperSlicer #2841: Cannot reposition object on plater

tasmaine, 2022-05-26:
### Version
 2.3.57.11

### Operating system type + version
Windows 10 Home v 21H2 (64-bit) OS build 19044.1706

### Behavior
* _Issue:  Cannot reposition object on plater.
* _Steps needed to reproduce the problem_
*   Import STL file.
*   Select object, then click on Move Icon.  Drag object to desired position.  When mouse is released, object reverts to center of plater.
*  _Expected Results_: Object remains in changed position.
* _Actual Results_:  Object reverts to center of plater.
  * _Screenshots if useful_

_Is this a new feature request?_:  No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[Tabletstaender.zip](https://github.com/supermerill/SuperSlicer/files/8779303/Tabletstaender.zip)
_

tome9111991, 2022-05-26:
i can't reproduce

you have enable auto center patrts?

tasmaine, 2022-05-26:
Thanks for the very quick response!  I can’t find the auto center option to check its status, and didn’t know it existed.  Where is it?

tasmaine, 2022-05-26:
Sorry for my first reply.  I was responding to the e-mail and did not see your image on the website.  You are correct.  Auto Center was enabled.  I will close this.  

 

Thank you again!

tasmaine, 2022-05-26:
I am amazed at the quick, and correct response.  Thank you!

closed 2022-05-26 by tasmaine